ros-simulation/gazebo_ros_pkgs #289: Unable to visualize robotiq hand

TheDash, 2015-01-18:
Here's the gzserver.log:

[1421539319 933159493] ODE body for link [link] does not exist, GetGravityMode returns default of 0
[1421539319 933186699] ODE body for link [link] does not exist, GetKinematic returns default of 0
[1421539319 933193354] ODE body for link [link] does not exist, GetEnabled returns default of 1
[1421539319 933316327] ODE body for link [link] does not exist, unable to SetEnabled
[1421539320 93099352] ODE body for link [link] does not exist, unable to respond to OnPoseChange
[1421539320 93141137] ODE body for link [link] does not exist, unable to SetKinematic
[1421539320 93152461] ODE body for link [link] does not exist, unable to SetGravityMode
[1421539320 93170844] ODE body for link [link] does not exist, unable to SetLinearDamping
[1421539320 93179399] ODE body for link [link] does not exist, unable to SetAngularDamping
[1421539320 93197517] ODE body for link [link] does not exist, unable to respond to OnPoseChange
[1421539320 93259085] ODE body for link [link] does not exist, unable to place collision bodies in ODELink::Init
[1421539320 93264877] ODE body for link [link] does not exist, unable to UpdateMass
[1421539320 93268267] ODE body for link [link] does not exist, unable to set callbacks in ODELink::Init
[1421539320 93529585] World::Init
[1421539321 530199226] ODE body for link [link] does not exist, GetGravityMode returns default of 0
[1421539321 530225412] ODE body for link [link] does not exist, GetKinematic returns default of 0
[1421539321 530229409] ODE body for link [link] does not exist, GetEnabled returns default of 1
[1421539321 730529146] ODE body for link [link] does not exist, GetGravityMode returns default of 0
[1421539321 730550064] ODE body for link [link] does not exist, GetKinematic returns default of 0
[1421539321 730556700] ODE body for link [link] does not exist, GetEnabled returns default of 1
[1421539375 704562984] ODE body for link [__default__] does not exist, GetGravityMode returns default of 0
[1421539375 704594452] ODE body for link [__default__] does not exist, GetKinematic returns default of 0
[1421539375 704603515] ODE body for link [__default__] does not exist, GetEnabled returns default of 1
[1421539375 704641435] ODE body for link [link] does not exist, unable to SetEnabled
[1421539375 704651864] ODE body for link [__default__] does not exist, unable to SetEnabled
[1421539375 704696891] ODE body for link [__default__] does not exist, unable to respond to OnPoseChange

This is for Gazebo 1.9 Do you know of any fix/what is going on here?

TheDash, 2015-01-18:
Nevermind, looks like I am the fool here. Trying to call spawn_model with a .xacro file fails. As well, if the terminal gazebo is launched in is not sourced to the proper workspace the URDF file with package:// calls will be unable to include resources.

TheDash, 2015-01-18:
[1421539867 715103320] ODE body for link [link] does not exist, GetGravityMode returns default of 0
[1421539867 715129680] ODE body for link [link] does not exist, GetKinematic returns default of 0
[1421539867 715136044] ODE body for link [link] does not exist, GetEnabled returns default of 1
[1421539867 715249562] ODE body for link [link] does not exist, unable to SetEnabled
[1421539867 876989979] ODE body for link [link] does not exist, unable to respond to OnPoseChange
[1421539867 877032867] ODE body for link [link] does not exist, unable to SetKinematic
[1421539867 877044058] ODE body for link [link] does not exist, unable to SetGravityMode
[1421539867 877062120] ODE body for link [link] does not exist, unable to SetLinearDamping
[1421539867 877071176] ODE body for link [link] does not exist, unable to SetAngularDamping
[1421539867 877090012] ODE body for link [link] does not exist, unable to respond to OnPoseChange
[1421539867 877154247] ODE body for link [link] does not exist, unable to place collision bodies in ODELink::Init
[1421539867 877159662] ODE body for link [link] does not exist, unable to UpdateMass
[1421539867 877163120] ODE body for link [link] does not exist, unable to set callbacks in ODELink::Init
[1421539867 877414656] World::Init
[1421539869 313009290] ODE body for link [link] does not exist, GetGravityMode returns default of 0
[1421539869 313038878] ODE body for link [link] does not exist, GetKinematic returns default of 0
[1421539869 313043097] ODE body for link [link] does not exist, GetEnabled returns default of 1
[1421539869 513896984] ODE body for link [link] does not exist, GetGravityMode returns default of 0
[1421539869 513930227] ODE body for link [link] does not exist, GetKinematic returns default of 0
[1421539869 513936680] ODE body for link [link] does not exist, GetEnabled returns default of 1

this is the new error after trying it with a proper .urdf and the workspace sourced. The gazebo terminal spits out the URDF string and I'm unable to see any errors.

TheDash, 2015-01-18:
after increasing my terminal size in the terminal preferences to "unlimited" scrolling, I was able to see that the error was

Error [parser.cc:697] XML Element[use_parent_model_frame], child of element[axis] not defined in SDF. Ignoring.[axis]
Error [parser.cc:688] Error reading element <axis>
Error [parser.cc:688] Error reading element <joint>
Error [parser.cc:688] Error reading element <model>
Error [parser.cc:348] Unable to read element <sdf>

I should be able to fix it from here. Hopefully this can serve as documentation for some!
Error [parser.cc:299] parse as old deprecated model file failed.
Error [World.cc:1469] Unable to read sdf string[<!-

closed 2015-01-18 by TheDash

scpeters, 2015-01-18:
The `use_parent_model_frame` tag was added in gazebo3 to help resolve [gazebo issue 494](https://bitbucket.org/osrf/gazebo/issue/494/make-joint-axis-reference-frame-relative). I think the robotiq hand might require gazebo3 or higher.

TheDash, 2015-01-18:
I've removed it so that it can be used with gazebo 1.9. What are the repercussions?

scpeters, 2015-01-18:
[gazebo issue 494](https://bitbucket.org/osrf/gazebo/issue/494/make-joint-axis-reference-frame-relative) was related to a weird interpretation of the coordinate frame for expressing joint axes. It may work fine, but if not, the joint axes will be incorrect and links may rotate in the wrong direction.

TheDash, 2015-01-18:
Its okayi think, since the parameters of the tags I removed were set to
false.

scpeters, 2015-01-20:
Actually `false` the reasonable value, and `true` is the value for the legacy support. Hopefully it just works anyway.