ros-simulation/gazebo_ros_pkgs #304: [gazebo_plugins/gazebo_ros_bumper]  Multiple bumper sensors problem

cyborg-x1, 2015-03-17:
Hi I am trying to create multiple bumper sensors, but unfortunately I am always getting 
> Error [ContactManager.cc:271] Filter with the same name already exists!
for any sensor, except the first one and only the first one is working then.

I filed a question on Gazebo answers, but did not get an answer for days now:
http://answers.gazebosim.org/question/8193/multiple-bumper-sensors-filter-with-the-same-name/

Here is a direct link to my code:
https://github.com/cyborg-x1/rapid_robot_xacros

Is there a problem with the plugin or am I doing something wrong?

scpeters, 2015-03-17:
Sorry, I haven't been keeping up with gazebo answers in the past few days. This was [issue 960](https://bitbucket.org/osrf/gazebo/issue/960/allow-multiple-contactsensor-per-collision) in gazebo and was recently fixed in [gazebo pull request 1413](https://bitbucket.org/osrf/gazebo/pull-request/1413/allow-multiple-contact-sensors-per-link/diff) in the `gazebo_4.1` `gazebo5` and `default` branches. We haven't made a new patch release yet for this.

@j-rivero

j-rivero, 2015-03-17:
Let's put the buildfarm to work: https://bitbucket.org/osrf/gazebo/issue/1521/release-new-patched-versions-412-and-502

cyborg-x1, 2015-03-17:
Ah ok, nice, then we can close this one.

closed 2015-03-17 by cyborg-x1