ros-simulation/gazebo_ros_pkgs #312: [ gazebo_plugins] use sensor_msgs/Range message for bumper plugin

bit-pirate, 2015-04-10:
AFAIK [sensor_msgs/Range](http://docs.ros.org/api/sensor_msgs/html/msg/Range.html) is the defacto standard message type for range sensors (IR, ultrasonic) including fixed-range sensors such as bumpers.

Hence, i suggest we switch the message type used for the bumper sensor plugin to this message.

Any thoughts/comments?

mryellow, 2015-07-07:
Would make sense for my use-case.

Although I'm having other issues with `contact` type sensors crashing so not getting far enough for it to be something I'm concerned with yet.

muhrix, 2015-07-10:
Isn't it the case that a bumper sensor is conceptually different from (fixed-) range sensors, since it measures [contact] (http://docs.ros.org/indigo/api/gazebo_msgs/html/msg/ContactState.html) rather than distance?

I see a bumper more like a fixed-threshold force sensor, which state change is triggered upon contact.

mryellow, 2015-07-10:
Range is a little odd too as you need to know what would constitute the maximum force to give a relative reading.

That said, I'm stuck, falling off the bumper sensor and beginning to look at writing my own Gazebo plugin to get past it. 

I'm unable to get it to work for a start, crashes, which seems to be that it needs an entire collision chain back to world frame, something my robot won't be providing until later tf broadcasts. 

I'm not sure how bumpers are used calls for all of this contact information to be required. Given you need it to figure out how hard/where a bumper is contacted, but then you need to much context to get those  numbers.

I wish to know when a bumper, floating free from any links, touches something. 

Maybe Gazebo5 and Jade is the answer....

bit-pirate, 2015-07-11:
> Isn't it the case that a bumper sensor is conceptually different from (fixed-) range sensors, since it measures contact rather than distance?
> 
> I see a bumper more like a fixed-threshold force sensor, which state change is triggered upon contact.

True.

Also, the message description reads:

> Single range reading from an active ranger that emits energy and reports one range reading that is valid along an arc at the distance measured. 
[sensor_msgs/Range Message](http://docs.ros.org/api/sensor_msgs/html/msg/Range.html)

This is also not the case for a bumper.

Then I guess a proposal to add a bumper message to the sensor_msgs package would be the way to go. Unfortunately, I have no time to chase that at this moment.

@mryellow please open a separate issue regarding your problems with running the bumper plugin (or ask on Gazebo answers). There will be a higher change for answers, than posting your problems here.

mryellow, 2015-07-12:
I'm not fully across the code-base so I may be off track....

Not sure I really have an issue other than to comment on how the frame references work, combined with this issue, seems to indicate that contact sensors are only usable on fixed frame robots with Indigo/Gazebo2 (or those with a tf to world with collision). So a robot with a range "type" sensor in the link frame reference (ignoring that you need a maximum relative value for a moment), isn't something which could be done far as I can tell.

Best I can figure the only solution for working with the contact sensor is to upgrade to Jade and Gazebo5, then work from there. As frame references changed.

https://bitbucket.org/osrf/gazebo/issues/545/request-change-contact-reference-frame

https://bitbucket.org/osrf/gazebo/pull-request/355/bullet-contact-sensor/diff

bit-pirate, 2015-07-13:
> Not sure I really have an issue other than to comment on how the frame references work, combined with this issue, seems to indicate that contact sensors are only usable on fixed frame robots with Indigo/Gazebo2 (or those with a tf to world with collision).

Ha, now I remember this unintuitive annoyance.

> Best I can figure the only solution for working with the contact sensor is to upgrade to Jade and Gazebo5, then work from there. As frame references changed.
> 
> https://bitbucket.org/osrf/gazebo/issues/545/request-change-contact-reference-frame

I was the one posting this issue. :-) Glad to hear it was changed in Gazebo5. 

Btw, you can run Gazebo5 with Indigo as well. We are probably going to that, since we have to stick with Indigo.

In any case, this is all offtopic regarding this issue here. So, let's continue this discussion somewhere else. :-)