ros-simulation/gazebo_ros_pkgs #313: gazebo_ros_gpu_laser and cpu laser only write out zero intensities.

skohlbr, 2015-04-13:
The laser plugins (both gpu and normal versions) appear to always retrieve 0 intensities from their internally used sensors (see for instance [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/hydro-devel/gazebo_plugins/src/gazebo_ros_gpu_laser.cpp#L202)). This makes it basically impossible to use intensity filtering, as 0 intensity scan point are among those one will want to discard in many real-life settings. To minimize special treatment for simulation, the ability to set the plugin to fill in a default minimum intensity would be highly useful. There apparently already has been a [old implementation](https://github.com/pal-robotics-graveyard/reemc_standalone/blob/master/overlay/atlas_msgs/gazebo_ros_laser.cpp#L109) of this by PAL Robotics. Or is there some other solution to achieve setting intensity?

hsu, 2015-04-13:
You can set [laser_retro](http://www.sdformat.org/spec?ver=1.5&elem=collision#collision_laser_retro) for all the collision elements, or the [gazebo_ros_block_laser](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/hydro-devel/gazebo_plugins/src/gazebo_ros_block_laser.cpp#L122) has a similar (same) implementation as the PAL robotics plugin. But I agree having the same mechanism in gazebo_ros_laser or gazebo_ros_gpu_laser is nice.

skohlbr, 2015-07-03:
As I didn't want to dive too deep into gazebo plugin land to have a workable solution, I made a lightweight node that takes LaserScan messages, modifies intensity, and republishes the scan: https://github.com/team-vigir/vigir_lidar_proc/blob/master/vigir_lidar_intensity_modifier/src/laser_intensity_modifier_node.cpp

Not really a solution, but a workaround if all one wants is non-zero intensity data.