ros-simulation/gazebo_ros_pkgs #316: Add plugin that publishes /joint_states

gerkey, 2015-04-17:
It would be very handy to have a plugin that publishes /joint_states data on all (by default) or some (if configured) joints in a model.  At present, the only way that I found to make this happen without writing code was to add in the URDF a transmission for every joint, load the gazebo_ros_control plugin, and configure and load a JointStateController.  That works, but is very XML/YAML-intensive, and requires me to specify things like motor reductions for passive joints that I'm not actually controlling (but want to see the state of, as if I had an encoder attached to each).

130s, 2016-06-24:
https://github.com/ros-controls/ros_controllers/pull/181 may be a fix, if relying on `ros_control` is acceptable.

chapulina, 2018-08-29:
If I understand the request correctly, I think this is covered by the `gazebo_ros_joint_state_publisher` plugin, added in https://github.com/ros-simulation/gazebo_ros_pkgs/pull/141.

I'll close this issue but please reopen if I misunderstood the intent.

closed 2018-08-29 by chapulina