ros-simulation/gazebo_ros_pkgs #318: Jade Release

wjwwood, 2015-04-29:
Since we can't decide on whether or not to break up `gazebo_ros_control` (https://github.com/ros-simulation/gazebo_ros_pkgs/issues/179#issuecomment-97279479), I propose we release everything but `gazebo_ros_control` into jade for now.

I've already made a `jade-devel` branch on which I've updated the use of the `gazebo` rosdep key with the new `gazebo5` rosdep key: https://github.com/ros-simulation/gazebo_ros_pkgs/commit/d236df87a9215d5c41999e7bfe45d96b8f006bb2

This is necessary because on Trusty the `gazebo` rosdep key is already in reference to Gazebo2. And since Jade is on Trusty as well we need a different rosdep key for Gazebo5, hence `gazebo5`.

@scpeters or @j-rivero we need to get an entry in the rosdep database for `gazebo5` for Trusty, Utopic, and Vivid. I looked briefly at the available gazebo packages in Vivid and I don't really know which one it should point to, but we can also create more than just the `gazebo5` key in rosdep. We could instead create a matching set of keys for rosdep which mimic the ubuntu layout. Anyways, I'll have to wait to get the rosdep keys sorted out before I can release.

wjwwood, 2015-04-29:
Gazebo packages in Vivid: http://packages.ubuntu.com/search?keywords=gazebo&searchon=names&suite=all&section=all

j-rivero, 2015-04-29:
We usually recommend two ways of using gazebo5 packages:
 * Install the `gazebo5` package if you plan on using gazebo but don't make development on top of it.
 * Install the `libgazebo5-dev` package (`libgazebo-dev` is just a transitional package) if you plan on developing on top of gazebo.

Given the nature of ROS and rosdep, I would say that we want to go with `libgazebo5-dev`.

wjwwood, 2015-04-29:
Well, the particular use we are describing is for a package building a plugin. But it seems more than that, we should probably have `libgazebo5` and `libgazebo5-dev` so that we can properly model the build time vs runtime dependencies. So I'd say we want a few rosdep keys:

- rosdep -> ubuntu
- gazebo5 -> gazebo5
- libgazebo5 -> libgazebo5
- libgazebo5-dev -> libgazebo5-dev

scpeters, 2015-04-29:
I think `libgazebo5-dev` should be a `build_depend` and `gazebo5` should be `run_depend`, since the gazebo and gzserver executables are only present in the `gazebo5` package and not the `-dev` package. Both of these depend on `libgazebo5`, so I'm not sure we need an explicit key for that.

wjwwood, 2015-04-29:
Well, a package which provides just a plugin (doesn't run gazebo) would just need to build depend on `libgazebo5-dev` and run depend on `libgazebo5`. Then a package which uses that plugin might actually run gazebo in a launch file or something and that would depend on `gazebo5` and this plugin package.

In practice the packages which provide the plugins will likely also have examples and so having a run depend on `gazebo5` is probably needed anyways. So strictly speaking we don't need the rule, but I think it would be good to have for completeness.

Either way, I'll leave it to you guys to come up with the correct rosdep key and open a pull requests against ros/rosdistro.

Thanks.

j-rivero, 2015-04-29:
> Both of these depend on libgazebo5, so I'm not sure we need an explicit key for that.

+1, I can not imaging a valid use case right now, so we can go with only two.

scpeters, 2015-04-29:
According to @wjwwood 's comment, it is possible to run_depend on `libgazebo5` ([kobuki_gazebo_plugins](https://github.com/yujinrobot/kobuki_desktop/tree/indigo/kobuki_gazebo_plugins) for example), but I'm not sure if there's any advantage to the added complexity.

scpeters, 2015-04-29:
I just made a pull request for the rosdep keys (https://github.com/ros/rosdistro/issues/8243), and made the build_depends point to `libgazebo5-dev` instead of `gazebo5` (a2fb41c5db70d81baef6750f90e6023d5fe10c0c).

wjwwood, 2015-04-29:
Great, you might want to touch the change logs and then we'll need to do a release (bumping the minor version) and then do the release into Jade.

scpeters, 2015-04-29:
Changelogs in 193dcbb9ef8c69e807c82430165ef349e2f5f2bb

j-rivero, 2015-04-29:
> Well, a package which provides just a plugin (doesn't run gazebo) would just need to build depend on libgazebo5-dev and run depend on libgazebo5

I see the point, it is valid yes, but I agree with Steve, the number of use cases to use libgazebo5 instead of gazebo5 is really minimal and does not worth it to introduce complexity with another key. So my +1 to the current status.

I suppose that we can add some notes about the gazebo key migration (from just gazebo to gazebo5/libgazebo5-dev) so the people coming from Indigo know well what to do to change. Maybe it also interesting to mention our [Migration.md document](https://bitbucket.org/osrf/gazebo/src/26ea991509925a7d0a2d4dfcb64719b92a315ae5/Migration.md?at=default)

wjwwood, 2015-04-29:
> I suppose that we can add some notes about the gazebo key migration (from just gazebo to gazebo5/libgazebo5-dev) so the people coming from Indigo know well what to do to change. Maybe it also interesting to mention our Migration.md document

Please also put it or link to it in the Jade migration page: http://wiki.ros.org/jade/Migration

j-rivero, 2015-04-29:
> Please also put it or link to it in the Jade migration page: http://wiki.ros.org/jade/Migration

I've added some quick notes to the document. http://wiki.ros.org/jade/Migration#Gazebo_Simulator

wjwwood, 2015-04-30:
Thanks @j-rivero that looks awesome.

scpeters, 2015-04-30:
I'm going to update it since `libgazebo5-dev` actually doesn't depend on `gazebo5`. The `-dev` package has headers and development files, while the `gzserver`, `gzlient`, and `gazebo` executables are only in `gazebo5`.

scpeters, 2015-04-30:
I just updated the wiki page; please review: http://wiki.ros.org/jade/Migration#Gazebo_Simulator

j-rivero, 2015-04-30:
+1, thanks Steve

scpeters, 2015-04-30:
https://github.com/ros/rosdistro/pull/8259

closed 2015-04-30 by wjwwood