ros-simulation/gazebo_ros_pkgs #321: [gazebo_plugins] Build fail: error: ‘class gazebo::physics::Joint’ has no member named ‘SetParam’

ipa-fxm, 2015-04-29:
With today's updates on ```indigo-devel```, I get the following build error:

```
Scanning dependencies of target gazebo_ros_skid_steer_drive
[  9%] Building CXX object gazebo_ros_pkgs/gazebo_plugins/CMakeFiles/gazebo_ros_skid_steer_drive.dir/src/gazebo_ros_skid_steer_drive.cpp.o
/home/fxm/projects/indigo/control/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_skid_steer_drive.cpp: In member function ‘virtual void gazebo::GazeboRosSkidSteerDrive::Load(gazebo::physics::ModelPtr, sdf::ElementPtr)’:
/home/fxm/projects/indigo/control/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_skid_steer_drive.cpp:259:25: error: ‘class gazebo::physics::Joint’ has no member named ‘SetParam’
     joints[LEFT_FRONT]->SetParam("fmax", 0, torque);
                         ^
/home/fxm/projects/indigo/control/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_skid_steer_drive.cpp:260:26: error: ‘class gazebo::physics::Joint’ has no member named ‘SetParam’
     joints[RIGHT_FRONT]->SetParam("fmax", 0, torque);
                          ^
/home/fxm/projects/indigo/control/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_skid_steer_drive.cpp:261:24: error: ‘class gazebo::physics::Joint’ has no member named ‘SetParam’
     joints[LEFT_REAR]->SetParam("fmax", 0, torque);
                        ^
/home/fxm/projects/indigo/control/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_skid_steer_drive.cpp:262:25: error: ‘class gazebo::physics::Joint’ has no member named ‘SetParam’
     joints[RIGHT_REAR]->SetParam("fmax", 0, torque);
                         ^
/home/fxm/projects/indigo/control/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_skid_steer_drive.cpp: In member function ‘virtual void gazebo::GazeboRosSkidSteerDrive::UpdateChild()’:
/home/fxm/projects/indigo/control/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_skid_steer_drive.cpp:311:27: error: ‘class gazebo::physics::Joint’ has no member named ‘SetParam’
       joints[LEFT_FRONT]->SetParam("vel", 0, wheel_speed_[LEFT_FRONT] / (wheel_diameter_ / 2.0));
                           ^
/home/fxm/projects/indigo/control/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_skid_steer_drive.cpp:312:28: error: ‘class gazebo::physics::Joint’ has no member named ‘SetParam’
       joints[RIGHT_FRONT]->SetParam("vel", 0, wheel_speed_[RIGHT_FRONT] / (wheel_diameter_ / 2.0));
                            ^
/home/fxm/projects/indigo/control/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_skid_steer_drive.cpp:313:26: error: ‘class gazebo::physics::Joint’ has no member named ‘SetParam’
       joints[LEFT_REAR]->SetParam("vel", 0, wheel_speed_[LEFT_REAR] / (wheel_diameter_ / 2.0));
                          ^
/home/fxm/projects/indigo/control/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_skid_steer_drive.cpp:314:27: error: ‘class gazebo::physics::Joint’ has no member named ‘SetParam’
       joints[RIGHT_REAR]->SetParam("vel", 0, wheel_speed_[RIGHT_REAR] / (wheel_diameter_ / 2.0));
                           ^
make[2]: *** [gazebo_ros_pkgs/gazebo_plugins/CMakeFiles/gazebo_ros_skid_steer_drive.dir/src/gazebo_ros_skid_steer_drive.cpp.o] Error 1
make[1]: *** [gazebo_ros_pkgs/gazebo_plugins/CMakeFiles/gazebo_ros_skid_steer_drive.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
Invoking "make -j4 -l4" failed
```

scpeters, 2015-04-29:
My fault! I merged #315, which requires gazebo 3+; I'll add ifdef's for gazebo2

scpeters, 2015-04-29:
Should be fixed by #322, thanks for the report! Sorry for breaking it.

PR #322 (2015-04-29): Add ifdefs to fix build with gazebo2
It was broken by #315.
Fixes #321.
commit 42061ade (2015-04-29): Add ifdefs to fix build with gazebo2

It was broken by #315.
Fixes #321.
commit b5303e69 (2016-01-15): Use Joint::SetParam for joint velocity motors

Before gazebo5, Joint::SetVelocity and SetMaxForce
were used to set joint velocity motors.
The API has changed in gazebo5, to use Joint::SetParam
instead.
The functionality is still available through the SetParam API.

cherry-picked from indigo-devel

Add ifdefs to fix build with gazebo2

It was broken by #315.
Fixes #321.
closed 2015-04-29 by scpeters