ros-simulation/gazebo_ros_pkgs #323: gazebo_ros has undeclared run dependency on libgazebo5-dev

scpeters, 2015-05-01:
The jade version of `gazebo_ros` declares a [build dependency on libgazebo5-dev](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_ros/package.xml#L27) and a [run dependency on gazebo5](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_ros/package.xml#L40).

However, the [gazebo_ros/scripts](https://github.com/ros-simulation/gazebo_ros_pkgs/tree/jade-devel/gazebo_ros/scripts) are [using pkg-config](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_ros/scripts/gzserver#L21) to find the prefix for the gazebo setup.sh file. This implies a run dependency on the gazebo.pc file in `libgazebo5-dev`.

If you don't have `libgazebo5-dev` installed, this will cause simple roslaunch's of gazebo to fail with something like the following error message:
~~~
$ roslaunch gazebo_ros empty_world.launch
...
/opt/ros/jade/lib/gazebo_ros/gzclient: 17: .: Can't open /share/gazebo//setup.sh
/opt/ros/jade/lib/gazebo_ros/gzserver: 22: .: Can't open /share/gazebo//setup.sh
[gazebo-2] process has died [pid 15709, exit code 2, cmd /opt/ros/jade/lib/gazebo_ros/gzserver -e ode worlds/empty.world __name:=gazebo __log:=~/.ros/log/bb3dc582-f051-11e4-96c1-902b3456ea33/gazebo-2.log].
log file: ~/.ros/log/bb3dc582-f051-11e4-96c1-902b3456ea33/gazebo-2*.log
[gazebo_gui-3] process has died [pid 15719, exit code 2, cmd /opt/ros/jade/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=~/.ros/log/bb3dc582-f051-11e4-96c1-902b3456ea33/gazebo_gui-3.log].
log file: ~/.ros/log/bb3dc582-f051-11e4-96c1-902b3456ea33/gazebo_gui-3*.log
~~~

Installing `libgazebo5-dev` is currently a valid workaround.

@wjwwood

cmaestre, 2015-05-05:
Hello! While trying to install ```libgazebo5-dev``` I got a dependency with ```libogre-1.9-dev```:
```
The following packages have unmet dependencies.
 libgazebo5-dev : Depends: libogre-1.9-dev but it is not going to be installed
```
If I try to install it (```sudo apt-get install libogre-1.9-dev```):

```
The following packages will be REMOVED
  libogre-1.8-dev ros-jade-desktop
  ros-jade-desktop-full ros-jade-librviz-tutorial ros-jade-rqt-robot-plugins
  ros-jade-rqt-rviz ros-jade-rviz ros-jade-rviz-plugin-tutorials
  ros-jade-rviz-python-tutorial ros-jade-simulators ros-jade-urdf-tutorial
  ros-jade-visualization-tutorials ros-jade-viz
```
It looks like ```libogre-1.9-dev``` is not compatible with other features of Jade.

scpeters, 2015-05-05:
Thanks for the report. Which Ubuntu distro are you using?

cmaestre, 2015-05-06:
I am using Ubuntu 14.04.1 LTS

scpeters, 2015-05-06:
We had intended to link against libogre-1.8-dev on trusty, as [discussed here](https://github.com/ros-infrastructure/rep/pull/89#issuecomment-69232117), but the gazebo deb is incorrectly linking with version 1.9 as you noted. We are [working on building new gazebo debs](https://bitbucket.org/osrf/release-tools/pull-request/88/fix-trusty-dependencies-for-vivid-gazebo/diff).

cmaestre, 2015-05-12:
Hi! Are the new gazebo debs availables? Because it looks like this should be solved in builds 23 and 24, but when I try to install ```libgazebo5-dev``` I still get the same dependency. 

I am not sure of being using the good repo:

sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'

scpeters, 2015-05-12:
@tfoote have you pulled the trusty debs from May 6?

* http://build.osrfoundation.org/job/gazebo5_vivid-debian-debbuilder/23/
* http://build.osrfoundation.org/job/gazebo5_vivid-debian-debbuilder/24

tfoote, 2015-05-13:
I just tried importing them and they will not override the existing packages because they have the same filename. Can we get them with a debian increment on the build number?

scpeters, 2015-05-14:
Ok, I bumped the arch number in the following builds:

* http://build.osrfoundation.org/job/gazebo5_vivid-debian-debbuilder/25/
* http://build.osrfoundation.org/job/gazebo5_vivid-debian-debbuilder/26/

scpeters, 2015-05-14:
@cmaestre The new debs have been built, and it looks like they are already in the shadow-fixed repository. It should be working soon.

cmaestre, 2015-05-15:
I just installed the library from the shadow-fixed repository and it works! Thanks a lot for the fast answer. You are doing great with ROS.

closed 2015-07-17 by j-rivero

Raym0ndKwan, 2015-08-05:
Works nice for me. Cheers.

mechatron1991, 2015-12-19:
@scpeters I have similar issue. I am installing ros jade in debian jessie. I am using ros packages built for ubuntu vivid. I tried installing gazebo5 package via synaptic and this is the error that I got

gazebo5:
 Depends: libgazebo5 (=5.0.1-1~vivid) but it is not installable
 Depends: libsdformat2 but it is not going to be installed
 Depends: libspnav0 but it is not going to be installed
 Depends: gazebo5-common (=5.0.1-1~vivid) but it is not installable
 Recommends: gazebo5-plugin-base  but it is not installable

Any idea how to proceed ?

I am not aware if this is the right place to post it. Please correct me if I am wrong.

Thank you :)

scpeters, 2016-03-28:
@mechatron1991 sorry I didn't respond to this comment sooner. Is it still an issue for you?

I see errors like this sometimes when I have another version of gazebo and/or sdformat installed. I usually try to remove all gazebo and sdformat debs then trying to reinstall the desired ones. Let me know if you have this issue.

mechatron1991, 2016-04-06:
@scpeters Thank you for your reply. I still have this problem. The problem I have noticed so far is that gazebo package is not present in stable release(debian jessie) where as it is present in testing and unstable releases. I am working on installing gazebo from the source in debian jessie. Is it possible to make gazebo package available for debian jessie ?

scpeters, 2016-04-06:
I'll need to ask @j-rivero for help with that question about debian jessie.

mechatron1991, 2016-04-06:
@scpeters Thank you. Meanwhile, I had posted a question about installing gazebo from source in debian jessie in _gazebo answers_ website. I will update here if I find a solution to my problem.

j-rivero, 2016-04-07:
> The problem I have noticed so far is that gazebo package is not present in stable release(debian jessie) where as it is present in testing and unstable releases.

That is right, the package was not old enough nor fully clean of bugs to go into stable on time for jessie.

There is an effort for migrating ROS packages to Debian which probably can help you to bring gazebo binaries to jessie. https://wiki.debian.org/DebianScience/Robotics/ROS
Check for the `http://sir.upc.edu/debian-robotics jessie-robotics main`which is managed by Leo Palomo-Avellanade who works in debian-science and uses ROS.

mechatron1991, 2016-04-07:
@j-rivero That solves the problem that I have. Thank you.

jacquelinekay, 2016-04-25:
I noticed this issue in the package.xml; how will it affect the upgrade to Gazebo 7 in Xenial?

scpeters, 2016-04-26:
I think the kinetic branch will need `libgazebo7-dev` as a run-time dependency in the package.xml

PR #330 (2015-07-17): run_depend on libgazebo5-dev (#323)
Declare the dependency.
It can be fixed later if we don't want it.
Fixes #323.