ros-simulation/gazebo_ros_pkgs #327: Angular velocity offset when using differential drive plugin with odometry source "world"

AlexReimann, 2015-05-26:
If the parameter odometrySource is set to world (default, and also the example on the gazebo website), you sometimes get an proportional offset between the angular velocity command and the published angular velocity (odom).
As shown in [this video](https://youtu.be/6WTu-mJ8Ukc) it jumps at a certain angular velocity. On a different setup we also experience  having the proportional offset all the time.

[Maybe related](http://answers.ros.org/question/47612/simulating-small-differential-drive-robot-in-gazebo/?answer=209915#post-id-209915)

This only happen for angular velocity; linear is fine.
The mode with using "encoder" instead of "world" as odometry source works fine.

fyi @bit-pirate

AlexReimann, 2015-05-27:
Updated: Added that the same bug happens to linear velocity when using the other odometry source.

This is / was quite frustrating ._.

AlexReimann, 2015-05-27:
Branch with hack on which the odometry source setting encoder works: [branch](https://github.com/AlexReimann/gazebo_ros_pkgs/tree/diff_drive_hack)
The [dodgy commit](https://github.com/AlexReimann/gazebo_ros_pkgs/commit/9bb82124be2b7557f5a7c0d789f50916fa865ea0).
I __strongly__ discourage people from using this, as I don't actually know what the problem in the code really is (just made a copy & paste and seemed to work). I'm leaving this here to help anybody who wants to look into this

AlexReimann, 2015-05-28:
Maybe related:
I get random spikes for angular velocity from odom using encoder as odometry source. It is not dropping down to zero but even sometimes going negative as can be seen [here](https://youtu.be/zHuOzeaVhM4) at around 0:06. The azure (blue) colored line is the angular velocity.

maxbader, 2015-06-03:
hello AlexReimann

I wrote the encoder odometry part of that driver. Currently I am busy but I will get back to it.
Thanks for sharing your problem and batches.

Markus