ros-simulation/gazebo_ros_pkgs #333: Allow tf_prefix to be empty instead of defaulting to robot namespace

muhrix, 2015-07-02:
I believe we should be allowed to have empty tf_prefix values, rather than forcing it to the same as the robot namespace.

From my understanding, tf_prefix and robot namespace are fundamentally different concepts and they should happen to be the same by choice, not by default.
Also, tf_prefix has been deprecated (even though lots of people might still use it).

I simulate several robots in Gazebo under different namespaces, but I want one and only one TF tree (in my case, I modified the URDF so that the robot namespace is used as a prefix for each TF frame name, avoiding frame name conflicts across robots).

Please see my proposed changes in 052c7b7548e16dfa23878e9ec300752d11ece9b1 (I am only using camera and laser scanner at the moment, hence no changes to other plugins).
I can make a pull request if my approach is actually useful.

Thanks!

maxbader, 2015-07-02:
Hello Muhrix

I wrote some of the code which uses now the robots instance name as default prefix.
There where two reasons.
1.) if you use the tf_prefix param you have to write a xarco for every robot
2.) it is a bad practice not to use prefix on tf's
I have to check the code but I am shure it is currently not bug free and has space for enhancements. So activating the tf_prefix makes sense, I didn't know that it stop working (I am still on a very old gazebo ros version) but I would like to the the robot instance name as default to avoid problems later if one goes mutli robot.

Greetings

muhrix, 2015-07-03:
Hi @maxbader 

Thanks a lot for explaining the rationale of the changes!

Using the robot namespace as tf_prefix makes sense on pre-Hydro ROS, but tf_prefix is (in theory) unsupported now.

Regarding your reasons, I have some comments.
1. you do not necessarily need to have a xacro for each robot (I have one xacro and I can spawn several robots in Gazebo -- I'll explain how I do it below)
2. while I agree it is bad practice not to have a prefix on TF frames, using the unsupported tf_prefix is bad practice (see http://wiki.ros.org/tf2/Migration#Removal_of_support_for_tf_prefix)

Now, back to the xacro file and how I do it.
Imagine you have a launch file spawning 2 robots in Gazebo, each of which on its own namespace: `ns_1` and `ns_2`.

Every time I call the xacro.py script to set the `robot_description`, I pass some parameters to the xacro file, such as:
```xml
    <arg name="robot_tf_prefix" default="husky_fei"/>

    <param name="robot_description"
           command="$(find xacro)/xacro.py 
                    '$(find husky_fei_description)/urdf/husky_fei.urdf.xacro'
                    prefix:=$(arg robot_tf_prefix)
                    robotns:=$(arg robot_namespace)"/>
```
Then, I changed my robot URDF/xacro file to use that parameter for link and joint names, like this:
```xml
    <link name="$(arg prefix)base_footprint">
```
Notice that the prefix is appended to the name, rather than resolved to a different scope. The TF frame name will be `husky_fei_base_footprint` publish on `/tf` instead of `base_footprint` published on `/husky_fei/tf`.

We should be able to choose from:
1. same robot namespace and tf_prefix (if one still uses tf_prefix)
2. different robot namespace and tf_prefix
3. no robot namespace and a chosen tf_prefix
4. a robot namespace and no tf_prefix

At the moment the Gazebo plugins are deciding for us, and we only have options 1 and 2.
In my particular case, I would definitely like to use option 4, so I can have multiple robots under different namespaces, but sharing the same TF tree.

In the worst case scenario, we could have a parameter indicating whether we would like the tf_prefix to be the same as the robot namespace.

Does that make sense?

jvgomez, 2015-10-07:
As an additional point to this discussion, I wonder whether [this piece of code](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_camera_utils.cpp#L281-L296) is actually correct. If the IF statement in line 282 is true the prefix is preprended twice to the frame name.

T045T, 2015-11-05:
would using ```'/'``` as the tf_prefix solve this problem? it should be stripped out by the resolve call anyways :)

maxbader, 2015-11-06:
Hi

I updated my multi-robot demo for ros jade, it is currently located in my fork of the gazebo_ros_pkgs at github https://github.com/maxbader/gazebo_ros_pkgs 
one can start it with  roslaunch gazebo_plugins diff_drive_multi_robot.launch
It would be nice to get a nice working example using the camera, laser scan and the diff_drive plug-in.

Greetings

neel-stark, 2020-07-02:
Regarding the use libgazebo_ros_laser.so in a multi robot simulation. 
I am launching the nodes for each robot in their respective namespaces and also reading the robot.urdf.xacro into robot_description under the namespace of each robot. So as to differentiate between the coordinate frames of each robot, I am passing a prefix argument to the xacro which is prepended before each link name (as proposed by muhrix) to avoid writing hardcoded urdf files for each robot. I am following this method of prepending a tf_prefix before each link name since setting tf_prefix on the parameter server is deprecated by tf2. Issue arises when I am using the plugin, with a <robotNamespace> tag, the value of which is automatically being set to tf_prefix_ and prepended before the link name. Thus the link name is finally becoming robotNamespace/$(arg prefix)/linkName.
How do I work around this issue? Thanks in advance.