ros-simulation/gazebo_ros_pkgs #334: Ros MoveIt Planning Scene Plugin Compatibility in Indigo

nrontsis, 2015-07-09:
I am not sure if this is the right place to ask.

@jbohren is the Ros MoveIt! Planning Scene Plugin [(this)](https://github.com/jhu-lcsr-forks/gazebo_ros_pkgs/blob/hydro-devel/gazebo_plugins/include/gazebo_plugins/gazebo_ros_moveit_planning_scene.h) compatible with Indigo?

MoveIt! is dying when I am trying it, and I am wondering if there is a compatibility issue.

Thanks,

Nikitas

gavanderhoorn, 2015-07-14:
@nrontsis: please use the [appropriate tracker](https://github.com/ros-planning/moveit_ros/issues) for MoveIt related issues.

closed 2018-06-18 by ironmig

mehditlili, 2019-01-28:
@gavanderhoorn I don't think that tracker is the correct place to post such an issue. The plugin itself is programmed on an old Gazebo version and since then the API has changed. So I think it makes sense to reopen this issue until that plugin (which mysteriously disappears starting from indigo) is fixed for new ros versions. The changes that need to be done do not seem very complicated and I'll see what I can do.

gavanderhoorn, 2019-01-29:
@mehditlili: bit of a blast-from-the-past this.

mehditlili, 2019-01-29:
@gavanderhoorn Yes but is there any new feature in Moveit/Gazebo that makes this plugin irrelevant? I was playing around with a robotic arm in gazebo but moveit kept planning paths that collide with objects that are spawned manually after loading the world. Those objects do also not appear in rviz.

mehditlili, 2019-01-29:
Yes, you are right. But playing God is fun sometimes :p . I was just interested in seeing how the arm can operate and if it can correctly plan to avoid obstacles that I tell it about. The pointcloud plugin was my next step.
Thanks for the link I'll take a look.

gavanderhoorn, 2019-01-29:
I'm not saying the plugin discussed here doesn't have its uses btw. I was as mystified as you (probably) are as to why it vanished somewhere between Hydro and Indigo.