ros-simulation/gazebo_ros_pkgs #337: gazebo_ros_pkgs compilation error

rs1990, 2015-07-11:
Hello,

I keep getting this issue while trying to catkin_make it on indigo. Any help?

CMake Error at gazebo_ros_pkgs/gazebo_ros/CMakeLists.txt:26 (find_package):
  By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gazebo", but
  CMake did not find one.

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/maverick/indigo_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/maverick/indigo_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

scpeters, 2015-07-12:
Do you have a gazebo deb installed?

bgoldfai, 2015-07-15:
I had the same issue in Ubuntu 15.04 and Jade. I solved the error by installing libgazebo5-dev.

Thomas10111, 2016-02-11:
Installing  libgazebo6-dev solved it.  Thanks.
libgazebo6-dev was removed when it installed libogre-1.9.0.

closed 2016-02-25 by j-rivero

whyscience, 2017-03-18:
for gazebo7 it should be libgazebo7-dev

kamalpandey1993, 2018-01-10:
for `debian stretch armhf` `libtinyxml2-4.0.1 ` has regression issues see [https://answers.ros.org/question/278733/rospack-find-throws-exception-error/](url). So while removing `libtinyxml2-4.0.1` , `libgazebo*-dev` also gets removed. So before compiling `gazebo_ros` , recompile (clean `build` and` devel` )other packages which have dependency on `libtinyxml2` with a newer version of the library and then install `libgazebo-dev` and recompile freshly, Till the problem is fixed