ros-simulation/gazebo_ros_pkgs #341: gazebo_ros_control installed from .debs crashes with gazebo4 and gazebo5 on 14.04/indigo

skohlbr, 2015-08-05:
I can spawn robots using gazebo-ros-control with standard Gazebo (gazebo2) installed from .debs on my indigo dystem (14.04/AMD64), but when  I try installing ros-indigo-gazebo4-* or ros-indigo-gazebo5-* packages instead, Gazebo crashes with the following terminal output:
<pre>
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/indigo/lib/controller_manager/spawner", line 204, in main
    resp = switch_controller(loaded, [], 2)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
Segmentation fault (core dumped)
[INFO] [WallTime: 1438775870.398446] [0.000000] Shutting down spawner. Stopping and unloading controllers...
[gazebo-23] process has died [pid 18712, exit code 139, cmd /opt/ros/indigo/lib/gazebo_ros/gzserver /home/kohlbrecher/argo_repo2/src/argo_scenarios_gazebo/argo_umad_gazebo/worlds/umad_blender_200000_poly.world -u __name:=gazebo __log:=/home/kohlbrecher/.ros/log/70662e5a-3b65-11e5-8c0e-0090f5e666cf/gazebo-23.log].
log file: /home/kohlbrecher/.ros/log/70662e5a-3b65-11e5-8c0e-0090f5e666cf/gazebo-23*.log
</pre>

The same issue can be observed on multiple other machines.

adolfo-rt, 2015-08-05:
Indigo supports Gazebo 2.x, so it should not come as a surprise that it crashes when loaded from a gazebo with a major version bump. I haven't started looking into the Jade migration of `gazebo_ros_control`, so I can't provide pointers to what needs to be addressed.

skohlbr, 2015-08-05:
The problem is that this worked for months and then suddenly started crashing around 2 months ago, suddenly forcing switching between Gazebo versions for simulating different robot systems. I didn't have time to look into the issue either (and haven't tried Jade yet).

adolfo-rt, 2015-08-05:
The last release of `gazebo_ros_pkgs` on indigo was [2.4.8 in March 17, 2015](https://github.com/ros-simulation/gazebo_ros_pkgs/commit/02ada3edf6cba303a1892e2862181c1d722a4f3e). I imagine this is the version of your binaries. If a few months ago you were also up-to-date, it was likely 2.4.8 as well. What can you say about the version of Gazebo in your setup for the last months?.

skohlbr, 2015-08-05:
In that case it's probably a gazebo4 update that got released mid-May that started to make things fail (or at least revealed the issue, which probably I expect needs to be patched on the gazebo_ros_control side).

adolfo-rt, 2015-08-05:
Yes. Since compatibility with more recent versions of Gazebo will be addressed in the jade migration (Gazebo 5), can we close this issue?.

skohlbr, 2015-08-05:
Shouldn't we close it when that migration is working (and hopefully also working for indigo with recent gazebo versions) ? While I'm surprised that one posted about this before, I expect my colleagues and I are not the only ones encountering it. Closing suggests things are back to working to people glancing over the issue tracker, which is not the case. That's at least why I'd let it stay open, but feel free to close.

Are there already plans for when the Jade migration will be performed?

adolfo-rt, 2015-08-06:
> Shouldn't we close it when that migration is working (and hopefully also working for indigo with recent gazebo versions) ? 

This issue is not about the jade migration, but about an unsupported Gazebo + gazebo_ros_pkgs combination not working. There are no plans of supporting Gazebo 4 or 5 under indigo. Still if a patch that is also backwards compatible with Gazebo 2 is proposed, I'm happy to review and merge it.

> While I'm surprised that one posted about this before, I expect my colleagues and I are not the only ones encountering it. Closing suggests things are back to working to people glancing over the issue tracker, which is not the case. That's at least why I'd let it stay open, but feel free to close.

If you think it will get more visibility and inform others with similar questions, I'm fine with leaving it open.

> Are there already plans for when the Jade migration will be performed?

No. I'm trying to find some time to release `ros_control` and `ros_controllers` first.

skohlbr, 2015-08-19:
Ok, so the solution is actually straightforward: One just has to add the jade version of gazebo_ros_pkgs to the workspace (yes, also on Indigo) and compile from source. This makes gazebo come up without crashing.

As a minor caveat, it appears that controller parameters that worked smoothly in Gazebo2 cause quite a lot of vibration in Gazebo4 in the example model I used.

closed 2015-08-19 by skohlbr

adolfo-rt, 2015-08-19:
> As a minor caveat, it appears that controller parameters that worked smoothly in Gazebo2 cause quite a lot of vibration in Gazebo4 in the example model I used.

`<humor>`
Not only the robot, but the user. I shake every time I have to migrate to a new Gazebo major version, as it likely implies re-tuning all my controllers. Haha!.
`</humor>`

This is somewhat expected, since in the optimistic case simulation fidelity might have improved. One then has to tune controllers from a less realistic to a more realistic model of the world.