ros-simulation/gazebo_ros_pkgs #346: Can't launch node gzserver nor gzclient

sergibada, 2015-08-11:
Hello everybody!

I've just installed ROS Jade, which includes gazebo-5.0, in Ubuntu 14.04.2.
The thing is I am having some problems with Gazebo. I can start the GUI by introducing "gzserver" and "gzclient" in different terminals, however, if I introduce

roslaunch MYROBOT_gazebo MYROBOT_gazebo.launch
(as following this tutorial: http://gazebosim.org/tutorials/?tut=ros_roslaunch)

This is what I get:

... logging to /home/bada/.ros/log/e9d01456-403b-11e5-821c-0c607644cff2/roslaunch-hplinux-3212.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hplinux:56633/

SUMMARY
========

PARAMETERS
 * /rosdistro: jade
 * /rosversion: 1.11.13
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [3224]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e9d01456-403b-11e5-821c-0c607644cff2
process[rosout-1]: started with pid [3237]
started core service [/rosout]
ERROR: cannot launch node of type [gazebo_ros/gzserver]: can't locate node [gzserver] in package [gazebo_ros]
ERROR: cannot launch node of type [gazebo_ros/gzclient]: can't locate node [gzclient] in package [gazebo_ros]

I don't know how to solve this, so the node will be able to be located. I hope someone can help me!!

suryaprakaz, 2015-09-19:
I am currently experiencing the same problem as you do . If you had found a solution for it, please help me out

sergibada, 2015-09-19:
Hi!
What's exactly the problem? Have you tested gazebo with the examples of the
gazebosim web?

suryaprakaz, 2015-09-19:
I solved the problem thankfully. The problem being that I had accidentally cloned gazebo_ros repositories in my workspace causing a clash in package names. I was actually looking for the gazebo_control package which doesn't come with the original installation of ros jade.

sergibada, 2015-09-19:
Oh good then! As you say gazebo_ros package doesn't include gazebo_control.
In order to avoid these problems I switched to ROS Indigo. Good luck!

suryaprakaz, 2015-09-20:
A good choice actually. But Indigo wasn't available for 15.04 Vivid version of Ubuntu and I had no choice. And for some unknown reason my laptop wasn't able to install 14.04. Anyway, if you face such errors in the future with Jade remember to clone gazebo_ros_control to your catkin_ws before proceeding. Thank You!

j-rivero, 2016-04-27:
Thanks @sergibada for answering to the problem. Closing.

closed 2016-04-27 by j-rivero

roboticsai, 2016-09-02:
im also getting the same error, what should i do? im using ros indigo with gazebo 7.

anchuanxu, 2020-12-10:
> 我很庆幸地解决了这个问题。问题是我不小心在工作区中克隆了gazebo_ros存储库，从而导致程序包名称冲突。我实际上是在寻找gazebo_control软件包，该软件包不随ros jade的原始安装一起提供。
厉害 确实是这个原因，必须重新编译才行，把之前的build和lib都删除

magic5al, 2022-07-03:
> I solved the problem thankfully. The problem being that I had accidentally cloned gazebo_ros repositories in my workspace causing a clash in package names. I was actually looking for the gazebo_control package which doesn't come with the original installation of ros jade.

thank for your idea.I delet catkin_ws(my workspace) and then remake one and ·catkin_init_workspace·
Thank~！（sorry for my poor English）