ros-simulation/gazebo_ros_pkgs #347: rosservice delete_model doesn't fully delete the model.

BenBlumer, 2015-08-11:
If I spawn a resource-intensive model (e.g. the PR2) the memory used by Gazebo increases (about 45mb for the PR2). Upon deleting it, A) the memory is still used by Gazebo and B) spawning a model with the same name doesn't work properly.

This is a problem because: If a user needs to delete and spawn a model a few times, Gazebo starts using a lot of memory and running really slowly.

Ubuntu 14.04, Gazebo 5.1.0, ROS Indigo. 
To duplicate: 

Launch an emptyworld:
`roslaunch gazebo_ros empty_world.launch `

CD to the PR2 folder (this is my Gazeo Models location):
`cd ~/.gazebo/models/pr2`
(Note the memory being used by gzclient and gzserver using, e.g., Ubuntu's System Monitor)

Spawn a PR2
```
rosservice call /gazebo/spawn_sdf_model "model_name: 'pr'
model_xml: '`cat model.sdf`'
robot_namespace: 'joe'
initial_pose:
  position: {x: 0.0, y: 0.0, z: 0.0}
  orientation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}
reference_frame: ''" 
```
The PR2 appears. 
(Note the memory being used has increased)

**issue A**
Now delete the model:
`rosservice call /gazebo/delete_model "model_name: 'pr'"`
The PR2 disappears. 
(Note the memory being used hasn't decreased. If you spawn again with a different name, you get a larger increase in memory use.)

**Issue B**

Spawn the PR2 again __with the same name__ 
```
rosservice call /gazebo/spawn_sdf_model "model_name: 'pr'
model_xml: '`cat model.sdf`'
robot_namespace: 'joe'
initial_pose:
  position: {x: 0.0, y: 0.0, z: 0.0}
  orientation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}
reference_frame: ''" 
```
Memory use increases. You can see the PR2's blue "sensor lines", but the model doesn't appear.

BenBlumer, 2015-08-27:
Anyone have a workaround for this problem?

hsu, 2015-08-27:
may be related to [this issue in gazebo](https://bitbucket.org/osrf/gazebo/issues/1702)

BenBlumer, 2015-08-27:
Seems likely. I used a fix that's similar to yours for respawning the models -- adding a suffix to "rename" them. Though I did it on the user side.  It was only once my simulations slowed down by a factor of 100x that I realized I needed a proper delete :)

Thanks for looking into this, Hsu!