ros-simulation/gazebo_ros_pkgs #351: gazebo_ros_pkgs for gazebo6

huang0171, 2015-08-14:
Does this package support gazebo6?

I use Ubuntu 14.04; ROS Indigo; and gazebo 6.

But the package gazebo_ros_pkgs doesn't work.

It shows something like this :

Could not find any path, checked the following paths:
/home/huang/catkin_ws2/devel/etc/libgazebo_ros_paths_plugin.so
/home/huang/catkin_ws2/devel/include/libgazebo_ros_paths_plugin.so
/home/huang/catkin_ws2/devel/lib/libgazebo_ros_paths_plugin.so
/home/huang/catkin_ws2/devel/libexec/libgazebo_ros_paths_plugin.so
/home/huang/catkin_ws2/devel/share/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/etc/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/include/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/lib/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/lib/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/libexec/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/share/libgazebo_ros_paths_plugin.so
Could not find any path, checked the following paths:
/home/huang/catkin_ws2/devel/etc/libgazebo_ros_api_plugin.so
/home/huang/catkin_ws2/devel/include/libgazebo_ros_api_plugin.so
/home/huang/catkin_ws2/devel/lib/libgazebo_ros_api_plugin.so
/home/huang/catkin_ws2/devel/libexec/libgazebo_ros_api_plugin.so
/home/huang/catkin_ws2/devel/share/libgazebo_ros_api_plugin.so
/opt/ros/indigo/etc/libgazebo_ros_api_plugin.so
/opt/ros/indigo/include/libgazebo_ros_api_plugin.so
/opt/ros/indigo/lib/libgazebo_ros_api_plugin.so
/opt/ros/indigo/lib/libgazebo_ros_api_plugin.so
/opt/ros/indigo/libexec/libgazebo_ros_api_plugin.so
/opt/ros/indigo/share/libgazebo_ros_api_plugin.so
Could not find any path, checked the following paths:
/home/huang/catkin_ws2/devel/etc/libgazebo_ros_paths_plugin.so
/home/huang/catkin_ws2/devel/include/libgazebo_ros_paths_plugin.so
/home/huang/catkin_ws2/devel/lib/libgazebo_ros_paths_plugin.so
/home/huang/catkin_ws2/devel/libexec/libgazebo_ros_paths_plugin.so
/home/huang/catkin_ws2/devel/share/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/etc/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/include/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/lib/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/lib/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/libexec/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/share/libgazebo_ros_paths_plugin.so
/opt/ros/indigo/share/gazebo_ros_pkgs/gazebo_ros/scripts/gazebo: 25: .: Can't open /share/gazebo//setup.sh

skohlbr, 2015-10-12:
If you clone the jade branch jade-devel branch and build it as part of your workspace, things should work. I do so on a 14.04/64Bit machine running gazebo6 isntalled from OSRF's .deb repository.

scpeters, 2015-10-28:
We broke ABI with the gazebo 6.3 release, but fixed it with 6.4 (and now 6.5). Can you try `apt-get update` and `apt-get install libgazebo6-dev` and try again?

See also: https://bitbucket.org/osrf/gazebo/issues/1762/gazebo6-always-crashes-when-adding-a-ros

j-rivero, 2016-02-25:
This should be now fixed.

closed 2016-02-25 by j-rivero