ros-simulation/gazebo_ros_pkgs #355: Overriding environment variables from launchfile

CatherineH, 2015-09-03:
Currently, a launchfile such as this:

```
<?xml version="1.0"?>
<launch>
  <env name="GAZEBO_MODEL_PATH" value="$(find my_gazebo_worlds)" />
  <env name="GAZEBO_RESOURCE_PATH" value="$(find my_gazebo_worlds)" />
  <!-- Bring up the gazebo gui and an empty world -->
  <include file="$(find gazebo_ros)/launch/empty_world.launch" >
    <arg name="world_name" value="$(find my_gazebo_worlds)/worlds/myworld.world"/>
  </include>  
</launch>

```

will setup the GAZEBO_MODEL_PATH variable, but it will not set up the GAZEBO_RESOURCE_PATH variable. This is means that although I can share worlds and models in a ros package, I can't share materials.

I believe this is because GAZEBO_RESOURCE_PATH is overwritten from within Gazebo on launch.

I'm not sure if this is a gazebo_ros issue, or more of a gazebo issue.

scpeters, 2015-09-03:
Which version of gazebo are you using? This might be fixed in more recent versions, but I need to double-check.

CatherineH, 2015-09-03:
I have this issue in 2.2, I've also tried it in gazebo 6 and not seen the error, but I have other problems with gazebo 6 that prevent me from fully loading my world.

scpeters, 2015-09-03:
This was fixed upstream in gazebo4 in [pull request 1430](https://bitbucket.org/osrf/gazebo/pull-requests/1430/preserve-previous-gazebo_model_path-values/diff). I'll see if we can backport a fix to `gazebo_2.2`.

scpeters, 2015-09-03:
If you feel like a nasty hack, you can try to manually apply the patch from that pull request to the `/usr/share/gazebo/setup.sh` file.

This is necessary because that script is invoked by the `gazebo_ros/scripts`, such as [gzserver](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_ros/scripts/gzserver#L21-L22).

scpeters, 2015-09-03:
[upstream pull request](https://bitbucket.org/osrf/gazebo/pull-requests/1889/backport-fixes-to-setupsh-from-pull)

closed 2015-09-04 by CatherineH