ros-simulation/gazebo_ros_pkgs #356: Gazebo_ros spawnmodel for non-robots

CatherineH, 2015-09-07:
I want to procedurally generate environments. Is there a way to use spawnmodel on non-robot files, i.e., sdf files describing worlds or models?

When I attempt to call spawnmodel with the argument -sdf on a xacro-generated sdf string without a <robot> element, I get the following error:

```
Warning [parser.cc:478] XML Attribute[xmlns:xacro] in element[model] not defined in SDF, ignoring.
Error [parser.cc:697] XML Element[gazebo], child of element[model] not defined in SDF. Ignoring.[model]
Error [parser.cc:688] Error reading element <model>
Error [parser.cc:348] Unable to read element <sdf>
Error:   Could not find the 'robot' element in the xml file
         at line 81 in /build/buildd/urdfdom-0.2.10+dfsg/urdf_parser/src/model.cpp
Error [parser_urdf.cc:2608] Unable to call parseURDF on robot model
Error [parser.cc:299] parse as old deprecated model file failed.
```

keerthanamanivannan, 2017-03-23:
Did you figure out why we get this error and how to fix it?

ironmig, 2018-06-15:
Just confirmed spawn model works with non-robot models on ros kinetic (gazebo 7). 
I ran
```rosrun gazebo_ros spawn_model -sdf -file checkerboard_plane.sdf -model checkerboard_plane```
with the [checkboard plane found on models.gazebo.org](http://models.gazebosim.org/checkerboard_plane/). Worked as expected. Closing as this is an old issue with no feedback or reproducibility

closed 2018-06-15 by ironmig