ros-simulation/gazebo_ros_pkgs #358: gazebo_ros_api_plugin.cpp unconditionally sets global param /use_sim_time

airballking, 2015-09-23:
Hi,

I'd like to use gazebo as a planning engine, i.e. not as a standalone simulator, on a real robot. Unfortunately, gazebo_ros_api_plugin.cpp unconditionally sets the global parameter ```/use_sim_time``` to ```true```:

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/indigo-devel/gazebo_ros/src/gazebo_ros_api_plugin.cpp#L482

Would it be possible to remove this line? The launch-file ```empty_world.launch``` already shows how to correctly set this parameter from the launch-file:

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/indigo-devel/gazebo_ros/launch/empty_world.launch#L15

Best,
Georg.

P.S.: I am working on ROS Indigo with Gazebo 2.2 and ```gazebo_ros_pkgs``` installed from debs.

ironmig, 2018-06-15:
A PR removing this line likely wouldn't be approved because it could break current users who assume the sim time will be set by the plugin. However, a parameter to optionally not to do this (defaulting to current behavior) would be useful in these situations

ironmig, 2018-06-15:
Actually, it looks like this behavior was fixed in #606. Issue can be closed @scpeters

airballking, 2018-06-15:
@ironmig Thanks for pointing this out! It's been a long time since I reported this, and I cannot tell how I fixed it at the time. But the fix you pointed out looks indeed useful. I'm closing this.

closed 2018-06-15 by airballking