ros-simulation/gazebo_ros_pkgs #360: [gazebo_ros] The gazebo_ros package needs to declare a run-time dependency on libgazebo5-dev

jim-rothrock, 2015-10-03:
The problem is that the package does not declare a runtime dependency on libgazebo5-dev
    which is the package that provides the .pc file.

    The quick workaround is to apt-get update && apt-get install -y libgazebo5-dev

    Please ticket this for gazebo_ros since using the pc file makes libgazebo5-dev a missing runtime 
    dependency. 

See http://answers.ros.org/question/217927/problem-with-running-gazebo-in-a-docker-container/?answer=218325#post-id-218325 for details.

scpeters, 2015-10-03:
Hm...this was previously posted as #323, which I thought we had fixed in #330. Which version of `gazebo_ros_pkgs` are you using? This should be in the 2.5.1 release for jade.

jim-rothrock, 2015-10-03:
I am using the version of gazebo_ros_pkgs in osrf/ros:jade-desktop-full from hub.docker.com, which is 2.5.0. It was my understanding that new Docker images are built automatically when new versions of ROS are released.

scpeters, 2015-10-03:
@ruffsl when do docker images get rebuilt?

ruffsl, 2015-10-08:
@scpeters , sorry for the late response! 
The Official Docker Hub ROS repo is rebuilt anytime it's parent image (the official Ubuntu image) is rebuilt, or whenever a pull request for its images are merged into the Official Docker Hub index. The [`osrf/ros`](https://hub.docker.com/r/osrf/ros/) Docker Hub repo is an automated build repo, and it will also trigger a rebuild whenever its parent image (the official ROS image) is rebuilt, or whenever we commit to the master branch of the [`osrf/docker_images`](https://github.com/osrf/docker_images) github repo. We can also manually trigger a rebuild if necessary.

Looking at the logs for the "Build Details" of `osrf/ros` repo, it looks like the last rebuild accrued about 10 days ago:
[image]

Inspecting a freshly pulled image returns similar info:
``` yaml
$ docker inspect osrf/ros:jade-desktop-full 
[
{
    "Id": "ccacfe3212912f07f6e8168c2e3ef087f1e9a7ca92ddff0c3c7308864ea06347",
    "Parent": "940ecf3dd8eed6dafa41c7d46f772a0fb056e4b6acb99bb281e9d91789bca76b",
    "Comment": "",
    "Created": "2015-09-29T00:55:50.731782013Z",
    "Container": "5f5886bfcc423b839a40fb45d1b9cdd33ea974e90212bd93fe4b389b672b91da",
    "ContainerConfig": {
        "Hostname": "e611e15f9c9d",
        "Domainname": "",
        "User": "",
        "AttachStdin": false,
        "AttachStdout": false,
        "AttachStderr": false,
        "ExposedPorts": null,
        "PublishService": "",
        "Tty": false,
        "OpenStdin": false,
        "StdinOnce": false,
        "Env": [
            "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
            "LANG=en_US.UTF-8",
            "ROS_DISTRO=jade"
        ],
        "Cmd": [
            "/bin/sh",
            "-c",
            "apt-get update \u0026\u0026 apt-get install -y     ros-jade-desktop-full=1.2.0-0*     \u0026\u0026 rm -rf /var/lib/apt/lists/*"
        ],
        "Image": "940ecf3dd8eed6dafa41c7d46f772a0fb056e4b6acb99bb281e9d91789bca76b",
        "Volumes": null,
        "VolumeDriver": "",
        "WorkingDir": "",
        "Entrypoint": [
            "/ros_entrypoint.sh"
        ],
        "NetworkDisabled": false,
        "MacAddress": "",
        "OnBuild": [],
        "Labels": {}
    },
    "DockerVersion": "1.8.1",
    "Author": "Dirk Thomas dthomas+buildfarm@osrfoundation.org",
    "Config": {
        "Hostname": "e611e15f9c9d",
        "Domainname": "",
        "User": "",
        "AttachStdin": false,
        "AttachStdout": false,
        "AttachStderr": false,
        "ExposedPorts": null,
        "PublishService": "",
        "Tty": false,
        "OpenStdin": false,
        "StdinOnce": false,
        "Env": [
            "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
            "LANG=en_US.UTF-8",
            "ROS_DISTRO=jade"
        ],
        "Cmd": [
            "bash"
        ],
        "Image": "940ecf3dd8eed6dafa41c7d46f772a0fb056e4b6acb99bb281e9d91789bca76b",
        "Volumes": null,
        "VolumeDriver": "",
        "WorkingDir": "",
        "Entrypoint": [
            "/ros_entrypoint.sh"
        ],
        "NetworkDisabled": false,
        "MacAddress": "",
        "OnBuild": [],
        "Labels": {}
    },
    "Architecture": "amd64",
    "Os": "linux",
    "Size": 826048741,
    "VirtualSize": 2438684361,
    "GraphDriver": {
        "Name": "aufs",
        "Data": null
    }
}
]
```

I'm still seeing the error as in the ros answers post, but am unable to get @tfoote workaround to fix the issue. Is there a package version we need to bump in order to break the docker build cache in the servers?

ruffsl, 2015-10-08:
[Update]
Just tried building the images locally using the `--no-cache` argument to discard the prior cache to build the `ros:jade-ros-core` and then simply used the Makefile in the osrf/docker_images repo to build the rest of the jade images. Then tried the same launch command, but still got the error. :anger:

scpeters, 2015-10-08:
Could it be the version specified in the `apt-get install` command (`ros-jade-desktop-full=1.2.0-0*`)?

vineethravulapalli, 2016-02-10:
# Hi there,
I had tried to install libagazebo5. Everything goes fine but at the end I get the following messages and the process terminates:
**"vineeth@vineeth-HP-Pavilion-15-Notebook-PC:~$ sudo apt-get install -y libgazebo5-dev
[sudo] password for vineeth: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgazebo5-dev : Depends: libsdformat2-dev (>= 2.1.0) but it is not going to be installed
                  Depends: libgazebo5 (= 5.1.0-1~trusty) but it is not going to be installed
                  Depends: gazebo5-plugin-base (= 5.1.0-1~trusty)
E: Unable to correct problems, you have held broken packages."
**
Can anyone please help me with this issue.