ros-simulation/gazebo_ros_pkgs #361: [gazebo_ros] spawning too slow if lot of resources in URDF

guihomework, 2015-10-05:
We developed a tool that reads some extra tags in a URDF file. These tags are not valid URDF tags nor valid SDF tags, just additional XML tags that gazebo should ignore.
Unfortunately inside these tags we re-used the similar syntax for resources since our tool uses rviz actually which can also resolve resources. They look something like this :
```
 <tactile topic="tactile">
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://some_package_name/some_path/taxel.stl"/>
      </geometry>
    </tactile>
```

We have 200 of similar entries in our URDF and due to these additional tags, gazebo_ros model spawner gets so slow, more than 30 seconds, that our controllers spawner timeout and models are not cleanly loaded now.

I guess the culprit is this :
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_ros/src/gazebo_ros_api_plugin.cpp#L553

any ```package://``` is resolved, with no cache, even for non SDF/URDF tags. This takes ages and is completely useless as gazebo will not read these resolved resources anyway.

I think adding caching to the resource resolver (our 200 tags have the same package source for instance) would help, even for other models with a lot of resources (typically a hand with a lot of links).
Not taking care of non URDF/SDF tags would be even better but more complex to implement in the current structure (separated resolving and urdf/sdf spawning)

Do you think that is an easy fix ? I could probably try to add a very basic package name cache in this local resolver if you think that is the way to go.

best
Guillaume

adolfo-rt, 2015-10-06:
> Do you think that is an easy fix ? I could probably try to add a very basic package name cache in this local resolver if you think that is the way to go.

I say go for it!. The solution is not complex and the potential for performance gains significant.

rhaschke, 2015-10-06:
@guihomework You should also change to search from the last replaced occurence, not [always from beginning](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/5d98dfe0b48f4aa2b9288be6d0cf7bfa3d15be38/gazebo_ros/src/gazebo_ros_api_plugin.cpp#L580). Simply replace `, 0` by `, pos1`.

guihomework, 2015-10-06:
@rhaschke well spotted, but was not the largest slow down. Package finding on NFS systems was killing us and appears less often with the implemented cache.

guihomework, 2015-11-25:
The promised fix was added in #362, could somebody review that ? We did not notice issues with this fix applied more than one month ago in our setup, only really faster loads.

scpeters, 2015-12-07:
merged in #362

commit 14838561 (2015-12-07): Merge pull request #362 from ubi-agni/indigo-devel

[gazebo_ros] fixes #361
closed 2015-12-07 by scpeters