ros-simulation/gazebo_ros_pkgs #367: URDF to SDF joint dynamics conversion

leegs, 2015-11-07:
Will you please add the ability to convert "springReference" and "springStiffness" attributes in the "dynamics" element for joints in URDF  the "spring_reference" and "spring_stiffness" elements in SDF when spawning a robot from URDF, please?  This will allow spring/damper to be added at joints without using controllers or creating a workflow to manually editing an SDF generated from "gz SDF -p" to have springs.  I would GREATLY appreciate!  I believe this one additional feature would greatly facilitate and simplify building URDF robot models that simulate well.

Thank you.

scpeters, 2015-12-07:
@hsu is this working with the latest version of sdformat3?

ycheng517, 2016-07-25:
@hsu just want to follow up on this issue.

tomatpasser, 2018-02-18:
This seems to be working in Gazebo8 atleast:

```
<gazebo reference="myJoint">
  <implicitSpringDamper>true</implicitSpringDamper>
  <springStiffness>10000</springStiffness>
  <springReference>0.1</springReference>
</gazebo>
```

Gives me
```
<dynamics>
  <friction>0</friction>
  <spring_reference>0.1</spring_reference>
  <spring_stiffness>10000</spring_stiffness>
</dynamics>
```