ros-simulation/gazebo_ros_pkgs #378: [gazebo_ros_control] Jade release

vrabaud, 2016-01-10:
Thx !

j-rivero, 2016-02-25:
Ongoing, issue #395

littlejohnyang, 2016-03-10:
Hopefully this will be released soon, because without gazebo_ros_control I am not sure if I should use ROS jade and latest version of gazebo for my new project.

j-rivero, 2016-03-10:
Maybe @adolfo-rt can comment on this to help you with your plans

alextoind, 2016-03-10:
@littlejohnyang I don't know if it may help you, but I am working with ROS Jade and `gazebo_ros_control` compiled by source with almost no problem (at least until now).

ipa-fxm, 2016-04-02:
waiting for this, too!

littlejohnyang, 2016-04-05:
I ended up compiling from source code. So far seems to be no problem. Trying out MoveIt at the same time now. Hopefully there will be no problem integrating everything together.

j-rivero, 2016-04-11:
The 2.5.3 Jade version will include the gazebo_ros_control: ros/rosdistro#11044

Thanks for the feedback.

closed 2016-04-11 by j-rivero