ros-simulation/gazebo_ros_pkgs #382: gazebo and mavlink connection problem

JM-SNU, 2016-01-20:
I already install the gazebo6, pkg, mavros, ...

And I run 'roscore' and 'make posix_sitl_default gazebo', then I can see iris quadcopter

And then, I run roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557".

Then, I cannot receive imu data or local position data, and so on (all topics are unknown)

However, I already install all package about gazebo and mavros, and I integrate gazebo with ROS

Whaat is the problem?

( In gazebo terminal, it turns out that mavlink sim timeout for 10ms)

ffurrer, 2016-01-20:
I think the better place to post this issue is:
https://github.com/mavlink/mavros/issues

And it would help if you post the actual terminal output, or a way that people can reproduce this.

ffurrer, 2016-01-20:
This was reposted on [mavlink](https://github.com/mavlink/mavros/issues/476), so it can be closed here.

closed 2016-01-20 by JM-SNU