ros-simulation/gazebo_ros_pkgs #386: Running gazebo_ros with DART does not work

balakumar-s, 2016-02-02:
Running `rosrun gazebo_ros gazebo -e dart` throws an error ` get_physics_properties service call does not yet support physics engine [dart] ` and looking at gazebo_ros_api_plugin.cpp shows a TODO for dart. Is there another way to start gazebo inside ROS with DART? I can run `gazebo -e dart` and it works.

scpeters, 2016-02-02:
I am able to run `gazebo_ros` with `dart`. It does give that error message, but I think it is actually running.  Try inserting a model with joints and increasing the number of solver iterations to something large (like 1000). DART doesn't use that parameter, so if the real-time factor is unaffected, then it's likely you are using DART.

Though we should implement the `get_physics_properties` service at some point.

balakumar-s, 2016-02-02:
Thanks! Though it would be nice if Gazebo shows the physics engine being used.

closed 2016-02-02 by balakumar-s

scpeters, 2016-02-03:
That's a great idea. I filed a feature request in the gazebo issue tracker. Feel free to comment there or vote for the issue.

* https://bitbucket.org/osrf/gazebo/issues/1853/display-which-physics-engine-is-in-use-in

jacknlliu, 2018-04-02:
The dart can't work with `gazebo_ros_control`, and the robot can't keep its joint state in gazebo.

````
Ubuntu 16.04
ROS kinetic
gazebo 9 with dart
gazebo_ros_pkgs from source
````