ros-simulation/gazebo_ros_pkgs #388: Release request for Gazebo 7

bchretien, 2016-02-10:
Gazebo 7 was recently released, and it is now the default version on Arch Linux. Since the Gazebo-related packages are already patched for Gazebo 7, would it be possible to make a new release?

Thanks!

alextoind, 2016-02-22:
I would appreciate it too: when trying to install `ros-jade-gazebo7-pkgs` it says that it depends on `ros-jade-gazebo7-plugins` which cannot be installed yet.

v4hn, 2016-02-23:
:+1: Please release a new version of the repository into jade

scpeters, 2016-02-24:
@j-rivero will be doing this soon.

j-rivero, 2016-02-24:
The plan is as follow: 
 * implement any kind of testing on the gazebo_ros_pkgs repo.
 * release new version of gazebo_ros_pkgs into the usual ROS channels (needed for gazebo_ros_pkgs)
 * release gazebo7_ros_pkg (all the metadata is in place)

alextoind, 2016-02-24:
Thank you very much @j-rivero @scpeters.

If I may ask, is there an even roughly scheduled release date?

j-rivero, 2016-02-24:
> If I may ask, is there an even roughly scheduled release date?

I would like to have it done before the end of the week.

alextoind, 2016-02-24:
That's great! Thank you very much.

j-rivero, 2016-02-27:
> That's great! Thank you very much.

Here we go: https://groups.google.com/a/osrfoundation.org/forum/#!topic/gazebo/8g_bJX12pO0

The thanks go to the great @scpeters which maintain the patches needed to keep the repo working with every new release of gazebo.

closed 2016-02-27 by j-rivero

bchretien, 2016-02-27:
@j-rivero perfect, thanks!