ros-simulation/gazebo_ros_pkgs #398: How to add a sensor to measure the relative information (e.g. position, distance) between different robots?

libing64, 2016-02-27:
I want to add a sensor plugins to measure the relative information between different robots, is there any guidance or tutorial for this?

scpeters, 2016-02-29:
The following tutorial describes how to use some different sensors with `gazebo_ros`.  A ray sensor could find the distance to other objects:

http://gazebosim.org/tutorials?tut=ros_gzplugins

libing64, 2016-03-03:
@scpeters  Can you explain more clearly? The distance obtained from ray sensor is the distance to the world, but how to get the distance to other robot?
Do I need to get other robot's position?

libing64, 2016-03-11:
I have solved this problem, and add the wireless receiver sensor plugin to my project,
 https://github.com/libing64/hector_gazebo/tree/indigo-devel/hector_gazebo_plugins/include/hector_gazebo_plugins.

Thanks.

closed 2016-03-11 by libing64