ros-simulation/gazebo_ros_pkgs #407: The time stamp of multiple cameras are not synchronized?

libing64, 2016-03-15:
[image]

lucasw, 2016-03-20:
This looks related to #408 - can you cut and paste your sdf/urdf <sensor> in here?

libing64, 2016-03-20:
@lucasw  
[generic_stereo_camera.urdf.xacro.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/181317/generic_stereo_camera.urdf.xacro.txt)

libing64, 2016-03-20:
@lucasw The two image messages from the plugin are not synchronized.

[gazebo_ros_multicamera.cpp.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/181320/gazebo_ros_multicamera.cpp.txt)

lucasw, 2016-03-23:
The rosbag stores all the messages at the time they were received (and rqt_bag visualizes them that way), not the time of their timestamp (and not every message has a timestamp)- so if you look at the header/stamp from these messages it as rosbag plays them back you should see the same stamp (try playing it back super slow, so only one update every second or two), which means that a synchronizer message filter will do the right thing with them.

(So the timestamps are synchronized with each other, but they are both wrong according to #408, but I'll update my pr for that with updates to multicamera)

http://comments.gmane.org/gmane.science.robotics.ros.user/8049

libing64, 2016-03-23:
@lucasw I don't agree that rosbag would use so stupid method(use the received time) to store the message.
Why not use the time stamp of the message?

lucasw, 2016-03-23:
I'm not familiar with the underlying rosbag code, but there might be efficiency reasons for that as well (save everything to disk in order received, and play back in order from disk, and not look at timestamps).  

It would probably be possibly to write code to process a bag into a new bag that reorders based on timestamp if a message has a timestamp, or maybe have an optional flag added to rosbag to do it live.