ros-simulation/gazebo_ros_pkgs #408: [gazebo_plugins] camera sensor timestamp is one update old

lucasw, 2016-03-18:
The camera message is getting sent out with a timestamp that is actually the last timestamp.

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_camera.cpp#L91

I think these first two functions need to pass in cur_time, right now they use last_update_time by default: 

        this->PutCameraData(_image);
        this->PublishCameraInfo();
        this->last_update_time_ = cur_time;

I'll get a pr with a test together.

http://answers.ros.org/question/229387/gazebo-camera-timestamp-is-10update_rate-out-of-date/

lucasw, 2016-03-19:
Is there a reason updateRate exists, when the gazebo sensor update_rate already controls the rate of the sensor callback?

lucasw, 2016-03-19:
I have a test now, and it works for

    rostest gazebo_plugins camera.test
    catkin_make run_tests_gazebo_plugins_rostest_test_camera_camera.test

But `catkin_make run_tests_gazebo_plugins` runs my test, which initially reports success, but then the next test never starts and camera.test fails overall because it runs out of time.  Do I need to do something more to shut down the test?

janjachnik, 2016-10-07:
Nice work - I was having issues due to this bug and was about to fix it myself before I saw this. 

Note that the openni_kinect plugin also has the same problem.

lucasw, 2016-10-07:
Check out #410 (which is stuck in limbo because the test infrastructure can't properly run the test?  Also it has gained some merge conflicts...).  Maybe you could make the same fixes for the kinect plugin in an independent pull request (I would guess very similar code and tests would work).

helenol, 2016-12-12:
👍  Thanks for this! This issue is completely blocking for me, I'm so glad you already figured out the answer.

For anyone having trouble finding the diff from @lucasw's branch, here is the simple comparison:
https://github.com/ros-simulation/gazebo_ros_pkgs/compare/8a390af1e950fd417e3b7a6652d1e8c23ea929fd...lucasw:camera_timestamp_408

Main change is to swap from
`this->sensor_update_time_ = this->parentSensor->LastUpdateTime();`
to
`this->sensor_update_time_ = this->parentSensor->LastMeasurementTime();`

(and the Get() version of these for Gazebo versions before 7).

j-rivero, 2016-12-12:
Thanks for the comment Helen. Could you verify if the pr #410 fix the issue for you?

helenol, 2016-12-13:
Using openni_kinect plugin here, but making the same changes as in as in the depth camera in #410  fixed it there.

lucasw, 2016-12-21:
Above are the rebased commits to kinetic version of the branch https://github.com/ros-simulation/gazebo_ros_pkgs/compare/kinetic-devel...lucasw:camera_timestamp_408_kinetic_devel based off the latest kinetic-devel.

ironmig, 2018-06-15:
Fixed in above mentioned merged PRs

closed 2018-06-15 by ironmig