ros-simulation/gazebo_ros_pkgs #409: [gazebo_ros] More than 1 gzserver rostest results in timeout

lucasw, 2016-03-22:
I can create rostests that use gzserver that run fine when run individually using rostest or catkin_make run_tests and the individual test name, but trying two tests results in a timeout.

I'm guessing gazebo isn't getting shut down properly.

Example in https://github.com/lucasw/gazebo_ros_pkgs/tree/gazebo_ros_tests

```
$  catkin_make run_tests_gazebo_ros
####
#### Running command: "make run_tests_gazebo_ros -j4 -l4" in "/home/lucasw/catkin_ws/build"
####
-- run_tests.py: execute commands
-- run_tests.py: execute commands
  /opt/ros/jade/share/rostest/cmake/../../../bin/rostest --pkgdir=/home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros --package=gazebo_ros --results-filename test_gazebo_ros.xml --results-base-dir /home/lucasw/catkin_ws/build/test_results /home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/test/gazebo_ros.test 
  /opt/ros/jade/share/rostest/cmake/../../../bin/rostest --pkgdir=/home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros --package=gazebo_ros --results-filename test_gazebo_ros2.xml --results-base-dir /home/lucasw/catkin_ws/build/test_results /home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/test/gazebo_ros2.test 
INFO: cannot create a symlink to latest log directory: [Errno 2] No such file or directory: '/home/lucasw/.ros/latest'
... logging to /home/lucasw/.ros/log/rostest-glover-23010.log
... logging to /home/lucasw/.ros/log/rostest-glover-23011.log
[ROSUNIT] Outputting test results to /home/lucasw/catkin_ws/build/test_results/gazebo_ros/rostest-test_gazebo_ros2.xml
[ROSUNIT] Outputting test results to /home/lucasw/catkin_ws/build/test_results/gazebo_ros/rostest-test_gazebo_ros.xml
[ INFO] /gazebo /home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:155 : Finished loading Gazebo ROS API Plugin.
[ INFO] /gazebo /tmp/binarydeb/ros-jade-roscpp-1.11.16/src/libros/service.cpp:80 : waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] /gazebo /tmp/binarydeb/ros-jade-roscpp-1.11.16/src/libros/service.cpp:122 : waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] /gazebo /home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:1985 : Physics dynamic reconfigure ready.
testgazebo_ros ... ok

[ROSTEST]-----------------------------------------------------------------------

[gazebo_ros.rosunit-gazebo_ros/cameraSubscribeTest][passed]

SUMMARY
 * RESULT: SUCCESS
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 0

rostest log file is in /home/lucasw/.ros/log/rostest-glover-23010.log
-- run_tests.py: verify result "/home/lucasw/catkin_ws/build/test_results/gazebo_ros/rostest-test_gazebo_ros.xml"
Built target run_tests_gazebo_ros_rostest_test_gazebo_ros.test
testgazebo_ros2 ... ERROR!
ERROR: max time [10.0s] allotted for test [gazebo_ros2] of type [gazebo_ros/gazebo_ros-test]
  File "/usr/lib/python2.7/unittest/case.py", line 331, in run
    testMethod()
  File "/opt/ros/jade/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/jade/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/jade/lib/python2.7/dist-packages/roslaunch/launch.py", line 681, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

rostest log file is in /home/lucasw/.ros/log/rostest-glover-23011.log
-- run_tests.py: verify result "/home/lucasw/catkin_ws/build/test_results/gazebo_ros/rostest-test_gazebo_ros2.xml"
Built target run_tests_gazebo_ros_rostest_test_gazebo_ros2.test
Built target run_tests_gazebo_ros_rostest
Built target run_tests_gazebo_ros
```

```
<?xml version="1.0" encoding="utf-8"?>
<testsuite errors="1" failures="0" name="unittest.suite.TestSuite" tests="1" time="25.667">
  <testcase classname="rostest.runner.RosTest" name="testgazebo_ros2" time="25.6671">
    <error type="RLTestTimeoutException">max time [10.0s] allotted for test [gazebo_ros2] of type [gazebo_ros/gazebo_ros-test]
  File "/usr/lib/python2.7/unittest/case.py", line 331, in run
    testMethod()
  File "/opt/ros/jade/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/jade/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/jade/lib/python2.7/dist-packages/roslaunch/launch.py", line 681, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
    </error>
  </testcase>
  <system-out><![CDATA[[ROSTEST]setup[/home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/test/gazebo_ros2.test] run_id[d590c5de-efe1-11e5-98f7-00dbdf11ceac] starting
?[1mstarted roslaunch server http://127.0.0.1:44023/?[0m

SUMMARY
========

PARAMETERS
 * /rosdistro: jade
 * /rosversion: 1.11.16
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)

[ROSTEST]setup[/home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/test/gazebo_ros2.test] run_id[d590c5de-efe1-11e5-98f7-00dbdf11ceac] done
[ROSTEST]Running test [gazebo_ros2]
?[1mROS_MASTER_URI=http://127.0.0.1:59785/?[0m
?[1msetting /run_id to d590c5de-efe1-11e5-98f7-00dbdf11ceac?[0m
?[1mprocess[rosout-1]: started with pid [23053]?[0m
started core service [/rosout]
?[1mprocess[gazebo-2]: started with pid [23060]?[0m
[ROSTEST]removing previous test results file [/home/lucasw/catkin_ws/build/test_results/gazebo_ros/rosunit-gazebo_ros2.xml]
[ROSTEST]running test gazebo_ros2
?[1mprocess[gazebo_ros2-3]: started with pid [23072]?[0m
[ROSTEST]tearDown[/home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/test/gazebo_ros2.test]
shutting down processing monitor...
[gazebo_ros2-3] killing on exit
[rosout-1] killing on exit
... shutting down processing monitor complete
[ROSTEST]rostest teardown /home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/test/gazebo_ros2.test complete
]]></system-out>
  <system-err><![CDATA[?[31m[gazebo-2] process has died [pid 23060, exit code 255, cmd /home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -r /home/lucasw/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/test/empty.world __name:=gazebo __log:=/home/lucasw/.ros/log/d590c5de-efe1-11e5-98f7-00dbdf11ceac/gazebo-2.log].
log file: /home/lucasw/.ros/log/d590c5de-efe1-11e5-98f7-00dbdf11ceac/gazebo-2*.log?[0m
?[31m[gazebo_ros2-3] escalating to SIGTERM?[0m
]]></system-err>
</testsuite>
````

jacquelinekay, 2016-04-25:
@scpeters 

Interesting, could this be due to issues with running parallel tests (multiple simultaneous gzservers is not possible, I believe)

j-rivero, 2016-04-26:
Spawning different gzservers is possible although they will need different ports, I think that currently we are not at that point when running the test suite as it is now.

The workaround is to run tests one by one, by using the `j1` option.

scpeters, 2016-04-26:
The travis file was fixed with `-j1` on `indigo-devel` in b972a83ffdf0e988c8572a4a908c783d369db2fd from #353 

I'll cherry-pick it to jade

scpeters, 2016-04-26:
`-j1` added to travis.yml in jade in b14eedeb6be6222168e03a0a0cefb1eddab2d5e7

scpeters, 2016-04-26:
I think using `-j1` is the safest thing to do.

jacquelinekay, 2016-04-26:
Maybe we could put that in the README.md for this repository as a reminder for developers who run the tests locally?

scpeters, 2016-04-26:
See #440 for the update to the documentation.

scpeters, 2016-04-26:
I'm going to mark this closed by #440.

PR #440 (2016-04-26): Recommend -j1 when running tests.
Closes #409.
commit 1d3293ee (2016-04-26): Recommend -j1 when running tests.

Closes #409.
commit 069ef992 (2016-04-26): Recommend -j1 when running tests.

Closes #409.
closed 2016-04-26 by scpeters