ros-simulation/gazebo_ros_pkgs #415: Kinect plugin registred depth

bhack, 2016-03-31:
Is it possible to publish a registered depth topic in the [gazebo kinect plugin](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_openni_kinect.cpp) using code in [depth_image_proc](https://github.com/ros-perception/image_pipeline/blob/9164a7dc6edd6a29719759b0ac62ff9297b1484b/depth_image_proc/src/nodelets/register.cpp)? 
It could be useful to simulate on apps that consume the registered topic with the real device. 
/cc @vrabaud

bhack, 2016-04-05:
Also [depth_image_proc/register node](http://wiki.ros.org/depth_image_proc?distro=jade) seems that cannot work in the actual status.

bhack, 2016-04-06:
@vrabaud  Topic depth/camera_info is empty. It has a particular motivation? Is depth/image_raw already registred with rgb/image_raw in gazebo kinect plugin?

vrabaud, 2016-04-07:
That plugin should use image_transport to make sure camera_info is published and to take advantage of the compressed depth plugin. http://wiki.ros.org/image_transport/Tutorials/PublishingImages

bhack, 2016-04-07:
I see that there is also [gazebo_ros_depth_camera.cpp](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_depth_camera.cpp). Do you know why will have still two different plugins?

bhack, 2016-04-07:
/cc @wkentaro as one of the last commiter of the kinect one.

wkentaro, 2016-04-07:
Actually, I don't know about it, I'm not the maintainer of gazebo_plugins package. I contributed just to fix bugs.

I'm mainly use kinect2 and don't know much about depth registration used in openni_launch. 
The points published by the Kinect gazebo plugin are pointxyzrgb, and it means it is registered, isn't it?
What's the problem?

bhack, 2016-04-07:
There are three issues:
1. The first is that depth_registered is in a different format of depth_raw. Could be useful to have the same on real and simulated so apps could consume the same thing.
2. I don't know why we still have two plugin here: gazebo_ros_depth_camera and gazebo_ros_openni_kinect
3. camera/depth/camera_info it is empty

bhack, 2016-04-07:
Probably someone of the @ros-simulation team know a little bit the history of this plugin.