ros-simulation/gazebo_ros_pkgs #416: kinect doesn't work in complex world?

libing64, 2016-04-02:
I want to use the kinect plugin to generate some data to test the rgbdslam algorithm.
The plugin works well in some  worlds from [hector_gazebo](https://github.com/tu-darmstadt-ros-pkg/hector_gazebo) , however, it seems that the plugin doesn't work in complex world generated by myself (no rgb image and no depth is published) .
The world generated by myself is like this:
[image]

Can anyone provide some help?

libing64, 2016-04-03:
This maybe caused by some components, after removing some models in the world, kinect plugin works now.

closed 2016-04-03 by libing64

websailor, 2018-09-03:
I just use turtlebot3 simulate  project to to validate  laser sensors in scene maps. When the robot moves forward or backward, the laser data is displayed correctly in RViz. once the robot turns, the laser sensor data is seriously distorted.  I thought  to find out the raysensor  simalation   of  the gazebo raysorsenor plugin. Can anyone provide some help?  urdf file and map  pic is below!

[urdf.zip](https://github.com/ros-simulation/gazebo_ros_pkgs/files/2344112/urdf.zip)

[image]
[image]

chapulina, 2018-09-04:
@websailor , https://answers.ros.org is a better place for this kind of question. It looks like you're having a problem with your fixed frame.