ros-simulation/gazebo_ros_pkgs #417: [gazebo_plugins] libgazebo_ros_camera - some objects incorrectly occluded

xperroni, 2016-04-04:
Camera images published by `libgazebo_ros_camera` show certain objects incorrectly occluded by others, or even by the ground itself.

**Steps to reproduce**

2. Start `gazebo_ros` with the attached world file;
3. Open RQT and connect the image viewer to the `/camera/image_raw` topic;
4. On the Gazebo UI, check option _View->Transparent_;
5. Check the RQT image viewer again.

See the attached ZIP archive for the world file and complementary files.

[occlusion_bug_files.zip](https://github.com/ros-simulation/gazebo_ros_pkgs/files/201979/occlusion_bug_files.zip)

**Results**

- **Expected:** the image on RQT should show the FIRST 2015 trash can model in full, as it's standing in front of the camera;
- **Actual:** Only the top half of the trash can is visible. When switching the _Transparent_ option on Gazebo, it can be seen on RQT that the remaining of the trash can is standing "behind" the ground plane, which was erroneously rendered over it.

See attached images for actual results from both Gazebo and RQT.

I'm using Gazebo 7 with ROS Indigo on Kubuntu 14.04.1.

View of the scene from Gazebo:

[image]

Camera image preview as seen from inside Gazebo:

[image]

Camera image from RQT:

[image]

Camera image from RQT after the _Transparent_ option is selected on Gazebo:

[image]

skohlbr, 2016-06-21:
I believe the same issue is tracked on the Gazebo Bitbucket issue tracker here: https://bitbucket.org/osrf/gazebo/issues/1837/vmware-rendering-z-ordering-appears-random

xperroni, 2016-06-23:
Yes, it looks very much like my problem. Unfortunately it seems the only way to fix it is to recompile Gazebo with anti-aliasing disabled, though.

j-rivero, 2016-07-14:
Let's keep track of the problem in the [gazebo issue](https://bitbucket.org/osrf/gazebo/issues/1837/vmware-rendering-z-ordering-appears-random)