ros-simulation/gazebo_ros_pkgs #420: Error during run of ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release

gunshi, 2016-04-10:
I have been following the below link for instructions for installing ROS on OSX
[http://wiki.ros.org/indigo/Installation/OSX/Homebrew/Source](url)

When on running the command,

> rosdep install --from-paths src --ignore-src --rosdistro indigo -y

I got the error that many people have reported:
"could not find successful installation of gazebo" (even though I relinked it just to be sure)
 I did a brew install of gazebo(gazebo1), and added in  --skip-geys gazebo at the end and it ran successfully.

Upon further running:

>  ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release

The process stops with the error:

> [ 44%] Built target gazebo_os_joint_state_publisher
> [ 44%] Built target gazebo_os_block_laser
> make: *** [all] Error 2
> <== Failed to process package 'gazebo_plugins': 
>   Command '['/Users/gunshigupta/ros_catkin_ws/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2
> 
> Reproduce this error by running:
> ==> cd /Users/gunshigupta/ros_catkin_ws/build_isolated/gazebo_plugins && /Users/gunshigupta/ros_catkin_ws/install_isolated/env.sh make -j4 -l4
> 
> Command failed, exiting.

So when i run the command:

> cd /Users/gunshigupta/ros_catkin_ws/build_isolated/gazebo_plugins && /Users/gunshigupta/ros_catkin_ws/install_isolated/env.sh make -j4 -l4

to reproduce the error, I get the errors:

> In file included from /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/src/MultiCameraPlugin.cpp:19:
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/include/gazebo_plugins/MultiCameraPlugin.h:46:25: error: no
>       type named 'MultiCameraSensorPtr' in namespace 'gazebo::sensors'; did you mean 'MultiCameraSensor'?
>     protected: sensors::MultiCameraSensorPtr parentSensor;
>                ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
>                         MultiCameraSensor
> /usr/local/Cellar/gazebo1/1.9.7/include/gazebo-1.9/gazebo/sensors/MultiCameraSensor.hh:40:11: note: 'MultiCameraSensor' declared
>       here
>     class MultiCameraSensor : public Sensor
>           ^
> [ 47%] Linking CXX executable /Users/gunshigupta/ros_catkin_ws/devel_isolated/gazebo_plugins/lib/gazebo_plugins/pub_joint_trajectory_test
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/src/MultiCameraPlugin.cpp:33:22: error: no member named
>       'reset' in 'gazebo::sensors::MultiCameraSensor'
>   this->parentSensor.reset();
>   ~~~~~~~~~~~~~~~~~~ ^
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/src/MultiCameraPlugin.cpp:45:22: error: no viable overloaded
>       '='
>   this->parentSensor =
>   ~~~~~~~~~~~~~~~~~~ ^
> /usr/local/Cellar/gazebo1/1.9.7/include/gazebo-1.9/gazebo/sensors/MultiCameraSensor.hh:40:11: note: candidate function
>       (the implicit copy assignment operator) not viable: no known conversion from
>       'shared_ptr<gazebo::sensors::MultiCameraSensor>' to 'gazebo::sensors::MultiCameraSensor &' for 1st argument
>     class MultiCameraSensor : public Sensor
>           ^
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/src/MultiCameraPlugin.cpp:48:7: error: invalid argument type
>       'sensors::MultiCameraSensor' to unary expression
>   if (!this->parentSensor)
>       ^~~~~~~~~~~~~~~~~~~
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/src/MultiCameraPlugin.cpp:57:7: error: invalid argument type
>       'sensors::MultiCameraSensor' to unary expression
>   if (!this->parentSensor)
>       ^~~~~~~~~~~~~~~~~~~
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/src/MultiCameraPlugin.cpp:63:50: error: member reference type
>       'sensors::MultiCameraSensor' is not a pointer; maybe you meant to use '.'?
>   for (unsigned int i = 0; i < this->parentSensor->GetCameraCount(); ++i)
>                                ~~~~~~~~~~~~~~~~~~^~
>                                                  .
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/src/MultiCameraPlugin.cpp:65:46: error: member reference type
>       'sensors::MultiCameraSensor' is not a pointer; maybe you meant to use '.'?
>     this->camera.push_back(this->parentSensor->GetCamera(i));
>                            ~~~~~~~~~~~~~~~~~~^~
>                                              .
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/src/MultiCameraPlugin.cpp:73:48: error: member reference type
>       'sensors::MultiCameraSensor' is not a pointer; maybe you meant to use '.'?
>     std::string cameraName = this->parentSensor->GetCamera(i)->GetName();
>                              ~~~~~~~~~~~~~~~~~~^~
>                                                .
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/src/MultiCameraPlugin.cpp:91:21: error: member reference type
>       'sensors::MultiCameraSensor' is not a pointer; maybe you meant to use '.'?
>   this->parentSensor->SetActive(true);
>   ~~~~~~~~~~~~~~~~~~^~
>                     .
> 9 warnings and 9 errors generated.

Can anyone help me with what needs to be modified, (and if this is not the right page for this question, then where i should ask for help).
Thanks!

gunshi, 2016-04-20:
If there are any other commands/tests I need to run to give more detailed info, I'd be happy to do it.

gunshi, 2016-04-20:
I did not find "MultiCameraSensorPtr" defined in any namespace when I tried to grep for it in the gazebo-1.9 package.

scpeters, 2016-04-20:
indigo requires gazebo2

scpeters, 2016-04-20:
maybe the tutorial needs to be updated

gunshi, 2016-04-20:
Hi, i ran brew install gazebo2, brew uninstall gazebo, and brew link gazebo2.
I thought I should run "rosdep install --from-paths src --ignore-src --rosdistro indigo -y" just to be safe, and it said:

> executing command [brew install gazebo]
and proceeded to download gazebo1 from osrf/simulation

Is that a problem? Do i need to edit some files to tell them they need to look into gazebo2 and not gazebo1, or should i just add "--skip-keys gazebo" at the end, and run 

> ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release

scpeters, 2016-04-20:
`--skip-keys="gazebo"` is mentioned somewhere else on that page, so I would try that. We should probably fix those installation instructions.

gunshi, 2016-04-20:
Hi, I tried that, and I get the following result:

> <== Finished processing package [194 of 226]: 'tf'
> 
> ==> Processing catkin package: 'gazebo_ros'
> ==> Building with env: '/Users/gunshigupta/ros_catkin_ws/install_isolated/env.sh'
> Makefile exists, skipping explicit cmake invocation...
> ==> make cmake_check_build_system in '/Users/gunshigupta/ros_catkin_ws/build_isolated/gazebo_ros'
> ==> make -j4 -l4 in '/Users/gunshigupta/ros_catkin_ws/build_isolated/gazebo_ros'
> [ 20%] Built target gazebo_ros_gencfg
> Scanning dependencies of target gazebo_os_paths_plugin
> Scanning dependencies of target gazebo_os_api_plugin
> make[2]: *** No rule to make target `/usr/local/Cellar/gazebo1/1.9.7/lib/libgazebo_ccd.dylib', needed by `/Users/gunshigupta/ros_catkin_ws/devel_isolated/gazebo_ros/lib/libgazebo_ros_paths_plugin.dylib'.  Stop.
> make[2]: *** Waiting for unfinished jobs....
> [ 40%] Building CXX object CMakeFiles/gazebo_ros_paths_plugin.dir/src/gazebo_ros_paths_plugin.cpp.o
> make[2]: *** No rule to make target `/usr/local/Cellar/gazebo1/1.9.7/lib/libgazebo_ccd.dylib', needed by `/Users/gunshigupta/ros_catkin_ws/devel_isolated/gazebo_ros/lib/libgazebo_ros_api_plugin.dylib'.  Stop.
> make[2]: *** Waiting for unfinished jobs....
> [ 60%] Building CXX object CMakeFiles/gazebo_ros_api_plugin.dir/src/gazebo_ros_api_plugin.cpp.o
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_paths_plugin.cpp:22:10: fatal error: 
>       'gazebo/common/SystemPaths.hh' file not found
> #include <gazebo/common/SystemPaths.hh>
>          ^
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:23:10: fatal error: 
>       'gazebo/common/Events.hh' file not found
> #include <gazebo/common/Events.hh>
>          ^
> 1 error generated.
> make[2]: *** [CMakeFiles/gazebo_ros_paths_plugin.dir/src/gazebo_ros_paths_plugin.cpp.o] Error 1
> make[1]: *** [CMakeFiles/gazebo_ros_paths_plugin.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> 1 error generated.
> make[2]: *** [CMakeFiles/gazebo_ros_api_plugin.dir/src/gazebo_ros_api_plugin.cpp.o] Error 1
> make[1]: *** [CMakeFiles/gazebo_ros_api_plugin.dir/all] Error 2
> make: *** [all] Error 2
> <== Failed to process package 'gazebo_os': 

>   Command '['/Users/gunshigupta/ros_catkin_ws/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2
> 
> Reproduce this error by running:
> ==> cd /Users/gunshigupta/ros_catkin_ws/build_isolated/gazebo_ros && /Users/gunshigupta/ros_catkin_ws/install_isolated/env.sh make -j4 -l4
> 
> Command failed, exiting.

> yguns-MacBook-Pro:ros_catkin_ws gunshigupta$ cd /Users/gunshigupta/ros_catkin_ws/build_isolated/gazebo_ros && /Users/gunshigupta/ros_catkin_ws/install_isolated/env.sh make -j4 -l4
> [ 20%] Built target gazebo_ros_gencfg
> make[2]: *** No rule to make target `/usr/local/Cellar/gazebo1/1.9.7/lib/libgazebo_ccd.dylib', needed by `/Users/gunshigupta/ros_catkin_ws/devel_isolated/gazebo_ros/lib/libgazebo_ros_paths_plugin.dylib'.  Stop.
> make[2]: *** Waiting for unfinished jobs....
> [ 40%] Building CXX object CMakeFiles/gazebo_ros_paths_plugin.dir/src/gazebo_ros_paths_plugin.cpp.o
> make[2]: *** No rule to make target `/usr/local/Cellar/gazebo1/1.9.7/lib/libgazebo_ccd.dylib', needed by `/Users/gunshigupta/ros_catkin_ws/devel_isolated/gazebo_ros/lib/libgazebo_ros_api_plugin.dylib'.  Stop.
> make[2]: *** Waiting for unfinished jobs....
> [ 60%] Building CXX object CMakeFiles/gazebo_ros_api_plugin.dir/src/gazebo_ros_api_plugin.cpp.o
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_paths_plugin.cpp:22:10: fatal error: 
>       'gazebo/common/SystemPaths.hh' file not found
> #include <gazebo/common/SystemPaths.hh>
>          ^
> /Users/gunshigupta/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:23:10: fatal error: 
>       'gazebo/common/Events.hh' file not found
> #include <gazebo/common/Events.hh>
>          ^
> 1 error generated.
> make[2]: *** [CMakeFiles/gazebo_ros_paths_plugin.dir/src/gazebo_ros_paths_plugin.cpp.o] Error 1
> make[1]: *** [CMakeFiles/gazebo_ros_paths_plugin.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> 1 error generated.
> make[2]: *** [CMakeFiles/gazebo_ros_api_plugin.dir/src/gazebo_ros_api_plugin.cpp.o] Error 1
> make[1]: *** [CMakeFiles/gazebo_ros_api_plugin.dir/all] Error 2
> make: *** [all] Error 2

I don't think its picking up gazebo2.

gunshi, 2016-04-20:
FYI brew install of gazebo created folders gazebo-2.2 in lib and include folders under /usr/local.

scpeters, 2016-04-20:
Have you cleared the build folder and `brew remove gazebo1`? There shouldn't be any gazebo1 files installed

gunshi, 2016-04-21:
Just wanted to confirm, 
i had the following folders and files under ros_catkin_ws:

install_isolated 
devel_isolated 
build_isolated
src 
indigo-desktop-full-wet.rosinstall

from which i had to delete build_isolated, and install_isolated right?
Thanks.

ffurrer, 2016-04-21:
I recommend you take a look at the following instructions on how to get ROS working under OSX.

https://github.com/mikepurvis/ros-install-osx

gunshi, 2016-04-22:
Thanks for the tutorial!
I was able to compile successfully by removing the devel, build and install packages.

closed 2016-04-22 by gunshi