ros-simulation/gazebo_ros_pkgs #424: Gazebo y-right/z-up camera frame is inconsistent with rviz/opencv x-right/y-down frame

lucasw, 2016-04-21:
It looks like the gazebo camera frame convention is not the same as rviz and opencv, which the image below shows

[http://answers.ros.org/upfiles/14612622642337765.png](url)

 In opencv, z is pointing into the image (the blue axis), x is right (the red axis), and y is down (green axis), while in the gazebo camera x is pointing into the image and z is up, y is right which is similar to the robot convention of x being forward and z up.

The image above is using an rviz/Camera to overlay the purple grid on the frame generated from the gazebo camera plugin, instead of the grid overlaying properly on the ground and going to toward the horizon rviz thinks the camera is pointed at the ground.

This example is running the gazebo_ros_demos rrbot_gazebo and rrbot_control launch files, and using standard Ubuntu 14.04 + Jade packages.

lucasw, 2016-04-21:
It looks like the plugin will put `<frameName>` into the camera header frame- it should not be the same as the `gazebo reference` frame in order to avoid this, but the user has to generate that frame on their own (which is error prone)- why not have the plugin generate the correct frame, and even generate a warning or error if the user puts the same frame into gazebo reference and frameName?

felixvd, 2018-07-11:
Should this have been closed when the PR was merged?

TomShkurti, 2018-08-12:
I'm still seeing this issue in Kinetic. There is a "frameName" field in the camera sensor plugin that I can specify, but it is not subsequently making it to TF. The only frame being published is the link frame, and the camera produces an image wrt that frame with x-forward-z-up convention. For the time being using static_transform_publisher as lucasw suggests.