ros-simulation/gazebo_ros_pkgs #425: Kinetic release

jacquelinekay, 2016-04-25:
This will be a team effort with: @scpeters @hsu @wjwwood @hugomatic @chapulina @j-rivero 

Here are a few tasks that I can think of off the top of my head:

- [ ] Make kinetic-devel branch
- [ ] Update to Gazebo 7 (that's the default system version of Gazebo in Xenial, yes?)
- [ ] Test if there are build/runtime regressions for the combination of existing code/Gazebo 7/Xenial
- [ ] Check if pending PRs should go into the Kinetic release
- [ ] Make tags and bloom release

wjwwood, 2016-04-25:
Let me know how I can help. Not sure where I fit in.

Gazebo7 is the version for Kinetic (Wily + Xenial): http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021

scpeters, 2016-04-25:
Ok, the latest indigo release is [2.4.10](https://github.com/ros-simulation/gazebo_ros_pkgs/releases/tag/2.4.10), and there have been a [few pull requests merged since then](https://github.com/ros-simulation/gazebo_ros_pkgs/compare/2.4.10...indigo-devel).

Let's triage the open pull requests to see if anything needs to go into `indigo-devel`, then merge them forward to `jade-devel`, make new releases of indigo and jade, then rebase the kinetic branch off of the jade tag.

jacquelinekay, 2016-04-25:
If you merge a PR to Kinetic with new features/major behavioral changes, make sure to edit the Migration guide:

http://wiki.ros.org/kinetic/Migration

j-rivero, 2016-04-26:
> If you merge a PR to Kinetic with new features/major behavioral changes, make sure to edit the Migration guide:
> 
> http://wiki.ros.org/kinetic/Migration

Done:  http://wiki.ros.org/kinetic/Migration#Gazebo_Simulator

jacquelinekay, 2016-04-26:
#426 was merged (with the CATKIN_IGNORE in gazebo_ros_control) so if you rebase branches that target kinetic_devel, you should be able to use pull request testing with the ROS build farm to see if your code still builds on Xenial/with Gazebo 7. Let us know if you have questions about how to do that. Note that the Travis job will still fail for anything that targets kinetic_devel, #431 will remove it.

jacquelinekay, 2016-04-26:
actually Dirk just told me that the pull request tester automatically merges with the target branch before, so you might not even need to rebase before using PR testing

bmagyar, 2016-05-03:
Thought you guys may appreciate that `control_toolbox` is released as well as `ros_controllers` ;)
Let me know if there's something missing that I can help with.

jacquelinekay, 2016-05-03:
Yep, we've un-ignored gazebo_ros_control for Kinetic as of this morning: ros-simulation/gazebo_ros_pkgs#456

j-rivero, 2016-09-14:
Closing this issue since it served as tracker from the times of releasing the first kinetic packages.

closed 2016-09-14 by j-rivero