ros-simulation/gazebo_ros_pkgs #449: spawn_model package_to_model feature is brittle

scpeters, 2016-04-27:
The `spawn_model` command recently added a `package_to_model` option in #288 that converts `package://` to `model://` in the mesh `filename` attribute. It uses regular expressions, which is a brittle way to modify xml. Support for both double and single quotes are being added in #445 but it is still sensitive to the order of attributes, since the `<mesh>` element can also have a `scale` attribute (see [atlas.urdf](https://bitbucket.org/osrf/drcsim/src/583228eb/atlas_description/urdf/atlas.urdf#atlas.urdf-13) for an example), and I believe this regular expression wouldn't work if the attributes are rearranged.

Here's the different cases I can think of:

* Combinations of quotes with filename first:
~~~
<mesh filename="package://description/meshes/mesh.dae" scale="1 1 1" />
<mesh filename='package://description/meshes/mesh.dae' scale="1 1 1" />
<mesh filename="package://description/meshes/mesh.dae" scale='1 1 1' />
<mesh filename='package://description/meshes/mesh.dae' scale='1 1 1' />
~~~

* Combinations of quotes with scale first:
~~~
<mesh scale="1 1 1" filename="package://description/meshes/mesh.dae" />
<mesh scale="1 1 1" filename='package://description/meshes/mesh.dae' />
<mesh scale='1 1 1' filename="package://description/meshes/mesh.dae" />
<mesh scale='1 1 1' filename='package://description/meshes/mesh.dae' />
~~~

ironmig, 2018-06-18:
Confirmed does not work if other attributes (like scale) come before the filename attribute. I'm going to use python's xml library to parse the xml and replace the filename attributes. This has the added benefit of detecting if there is an xml error quickly on the client side before sending it off to gazebo

ironmig, 2018-07-06:
Fixed in #730, now uses python's xml library to find the filename attribute in any mesh tags, and url library to replace the protocol. Should be very robust

closed 2018-07-06 by ironmig