ros-simulation/gazebo_ros_pkgs #470: gazebo_ros_control missing from Jade

davetcoleman, 2016-06-28:
There is currently no debian available for ros-jade-gazebo-ros-control, as show on its [release status](http://repositories.ros.org/status_page/ros_jade_default.html?q=gazebo_ros_control). Why not? I know there have been ros_control release lags in the past, but ros_control has long been released into Jade. Surprisingly gazebo_ros_control is available in kinetic.

j-rivero, 2016-06-30:
I thought that https://github.com/ros/rosdistro/pull/11044 made them available from 2.5.3, but you are right I can not see them. @tfoote any idea about where should I look to find the build/upload log or how to trace the problem?

tfoote, 2016-06-30:
The debs exist and are in the testing repo as the link you pointed to shows. We have not synced jade in a long time due to the [several regressions](http://repositories.ros.org/status_page/ros_jade_default.html?q=REGRESSION) that have not been fixed. 

Before they are synced use the shadow repository. http://wiki.ros.org/ShadowRepository 

To request a sync please ping ros-release@lists.ros.org FYI @wjwwood

closed 2016-06-30 by tfoote