ros-simulation/gazebo_ros_pkgs #474: Release Request 2.5.6

furushchev, 2016-07-20:
Could you release new version for indigo / jade?
Almost 2 months have passed since the last release and there is important not yet released:
- https://github.com/ros-simulation/gazebo_ros_pkgs/commit/df82c78fb5ef20ea463dc8443b140161acee13da

j-rivero, 2016-07-21:
I released [2.4.11 for Indigo](https://github.com/ros/rosdistro/pull/12158) one week ago.
For jade, I'm waiting to merge https://github.com/ros-gbp/gazebo_ros_pkgs-release/pull/8. As soon as it is done, I will fire up bloom.

furushchev, 2016-07-21:
@j-rivero Thanks for information!

j-rivero, 2016-08-04:
After some conflicts with versioning, I just released 2.6.0 for Jade: https://github.com/ros/rosdistro/pull/12271 with the latest changes. That should be enough, feel free to reopen the issue if something is still needed.

Thanks.

closed 2016-08-04 by j-rivero

furushchev, 2016-08-10:
@j-rivero Thanks for informing!