ros-simulation/gazebo_ros_pkgs #475: Superfluous error messagy while calling /gazebo/get_world_properties

ipa-srd-rd, 2016-07-21:
When you call the `/gazebo/get_world_properties` service, a superfluous error message is printed to the screen:

Error [gazebo_ros_api_plugin.cpp:907] disablign rendering has not been implemented, rendering is always enabled

Despite the wrong wording, I'm not quite sure if it's necessary to print an error message without any check or failure. I hope you could fix this for ros indigo. Below the link to the codeline containing the error output.

[https://github.com/ros-simulation/gazebo_ros_pkgs/blob/1eff1d6f5ccd1b3d0a85cf50c97794471cd3aebe/gazebo_ros/src/gazebo_ros_api_plugin.cpp#L929](url)

mikaelarguedas, 2016-08-17:
+1
link update: https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_ros/src/gazebo_ros_api_plugin.cpp#L912

chapulina, 2019-12-26:
Removed on #747

closed 2019-12-26 by chapulina