ros-simulation/gazebo_ros_pkgs #478: GazeboRos constructor causes segfault

canalteks, 2016-07-27:
Dear developers,

I found that GazeboRos constructor for Sensor in "gazebo_ros_utils.h" causes segfault.
It's because that the member variable "rosnode_" was not initialized before using it.

- in GazeboRos( sensors::SensorPtr _parent, ... ) constructor.
- that calls ["GazeboRos::readCommonParameter()".](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/include/gazebo_plugins/gazebo_ros_utils.h#L173)
- the member function uses "rosnode_" variable to initialize ["tf_prefix_"](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_utils.cpp#L87) variable.
- but the "rosnode_" variable was not initialized in the GazeboRos( sensors ... ) constructor.

I hope, it could be helpful.

j-rivero, 2016-08-04:
Thanks for the report. Could you please attach an example code that triggers the segfault?

canalteks, 2016-08-04:
Thanks for your reply.

It could easily regenerate segfault just to call "[GazeboROS( sensors::SensorPtr, ... )](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/include/gazebo_plugins/gazebo_ros_utils.h#L156)".

Following code might be help.

````
class plugin_t : public gazebo::SensorPlugin
{
 void Load( gazebo::sensors::SensorPtr s, sdf::ElementPtr e );
 ...
}

void plugin_t::Load( gazebo::Sensors::SensorPtr s, sdf::ElementPtr  e )
{
 ...
 GazeboRos*  gaz_ros = new GazeboRos(s,e,"plugin_name");
 ...
}
````

To avoid the segfault, I've inserted the code before [at line 171 in gazebo_ros_util.h](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/include/gazebo_plugins/gazebo_ros_utils.h#L171).

````
 rosnode_ = boost::shared_ptr<ros::NodeHandle>( new ros::NodeHandle(namespace_));
````

Please see my previous comment, the reason why the segfault has occurred. 

My environment is
 - Debian jessie
 - ROS kinetic

jonbinney, 2016-11-30:
This definitely looks like a bug. Looks like one of the constructors for the `GazeboRos` initializes the node handle, but the other doesn't. I've seen periodic segfaults in gazebo when using with ROS, but if this is the cause I'm not sure why it doesn't happen every time.

scpeters, 2018-05-24:
Fixed in melodic by #718, but we should backport to lunar and kinetic too

canalteks, 2018-05-25:
@scpeters 
I appliciate your efforts. Thank you.

closed 2018-05-25 by canalteks

j-rivero, 2018-05-25:
Fix committed and propagated to all branches. Thanks everyone for the efforts.

PR #718 (2018-05-24): ROS UTILS: prevent segfault when using alternative GazeboRos constructor
Fix #478. This line was likely omitted by accident. When users use this constructor, rosnode_ is not initialized and the program will segfault when rosnode_ is used.
PR #722 (2018-05-25): ROS UTILS: prevent segfault when using alternative GazeboRos constructor (indigo-devel)
{ port of pull request #718 }
Fix #478. This line was likely omitted by accident. When users use this constructor, rosnode_ is not initialized and the program will segfault when rosnode_ is used.