ros-simulation/gazebo_ros_pkgs #484: [gazebo_plugins] RGB channel flipped with <image><format>R8G8B8</format></image> with openni_camera plugin

wkentaro, 2016-08-30:
Do you have idea how to fix this? Bug in gazebo library?

```
<image>
  <format>R8G8B8</format>
</image>
```
[image]

```
<image>
  <format>B8G8R8</format>
</image>
```
[image]

iche033, 2016-08-30:
[related issue](https://bitbucket.org/osrf/gazebo/issues/1644/rgb-channel-swapping-in-ros-visualization) in gazebo

scpeters, 2016-08-30:
@iche033 beat me to it, I was just about to link to that issue.

ironmig, 2018-06-29:
This can still be reproduced in kinetic/gazebo7 (and possibly other versions). Note that the point cloud from the openni_kinect plugin is generated by this plugin (not gazebo) based on the depth image. Looking at [the code](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_openni_kinect.cpp#L361), it is indeed hard coded to assume the image is RGB. I'll submit a PR to look at the image format and form the point cloud accordingly.

shmpwk, 2020-08-09:
Is this issue solved?
I'm having similar problem on `libgazebo_ros_depth_camera.so` on ROS melodic + gazebo-plugins `2.8.7`