ros-simulation/gazebo_ros_pkgs #485: ctrl-q doesn't tear down gzserver

jacquelinekay, 2016-08-30:
On 16.04 with the ros-kinetic-gazebo-ros debian, whenever I use CTRL-Q to exit Gazebo launched with `rosrun gazebo_ros gazebo`, I get the following error message:

```
/opt/ros/kinetic/lib/gazebo_ros/gazebo: 36: kill: invalid signal number or name: SIGINT
```

The gzserver process remains alive in the background and I have to kill it manually.

scpeters, 2016-08-30:
The signal names seem to have changed on some OS's. We fixed OSX in #364 but haven't fixed xenial yet. I suggested the following in https://github.com/ros-simulation/gazebo_ros_pkgs/pull/364#issuecomment-231419305 but I haven't tested it:

~~~
diff --git a/gazebo_ros/scripts/gazebo b/gazebo_ros/scripts/gazebo
index 392dac7..3edbf83 100755
--- a/gazebo_ros/scripts/gazebo
+++ b/gazebo_ros/scripts/gazebo
@@ -6,10 +6,8 @@ final="$@"
 final=`echo "$final" | sed 's/[^ ]*:=[^ ]* \?//g'`
 
 EXT=so
-SIGNAL=SIGINT
 if [ $(uname) = "Darwin" ]; then
     EXT=dylib
-    SIGNAL=INT
 fi
 
 # add ros path plugin if it does not already exist in the passed in arguments
@@ -40,4 +38,4 @@ GAZEBO_MASTER_URI="$desired_master_uri" gzserver $final &
 GAZEBO_MASTER_URI="$desired_master_uri" gzclient $client_final
 
 # Kill the server
-kill -s $SIGNAL $!
+kill -n 2 $!
~~~

scpeters, 2018-05-24:
fixed in #711 for lunar

closed 2018-05-25 by j-rivero