ros-simulation/gazebo_ros_pkgs #486: Remapping topics in gazebo nodes

jaredmoore, 2016-08-31:
I'm currently troubleshooting a problem with remapping the clock topic in a gazebo node.  A detailed explanation of the problem is here: http://answers.ros.org/question/242656/is-it-possible-to-remap-clock/

A synopsis is the following.  Ros Kinectic/Gazebo7.  Using `<remap>` either at launch file level, or within the `<node>` tags launching `gzserver` is failing.  Running it on a Ros Indigo/Gazebo2 VM allows the code to work.  Has there been a change to the remapping policy, solely within the gazebo nodes for this package?  

gazebo_only.launch:
```
<launch>

<!-- ********************* Begin Gazebo Defaults ******************************** -->

  <!-- these are the arguments you can pass this launch file, for example paused:=true -->
  <arg name="args" default="--verbose -r -e ode"/>
  <arg name="world" value="worlds/empty.world"/>
 
  <!-- set use_sim_time flag -->
  <!--<param name="/use_sim_time" value="true" />-->

<!-- ********************* End Gazebo Defaults ******************************** -->    

  <!-- start gazebo server-->
  <!--<env name="GAZEBO_MASTER_URI" value="http://localhost:11345"/>-->
  <node name="gazebo" pkg="gazebo_ros" type="gzserver" args="$(arg world) $(arg args)" respawn="false" output="screen">
    <remap from="rosout" to="rosout_remapped" />  
    <remap from="clock" to="clock_remapped" />
  </node>

</launch>

```

Test with:
1. roslaunch <package name> gazebo_only.launch

2. Open another terminal

3. source devel/setup.sh

4. rostopic list

Expected output:

`/clock` should be remapped to `/clock_remapped` and `/rosout` should be remapped to `/rosout_remapped`.

On my Kinectic/Gazebo7 VM this does not work correctly, but on an Indigo/Gazebo2 VM it does properly remap the topics.  Adding other nodes in either version gives them the remapped topics, but Gazebo still remains mapped to the `/clock` topic.

anthony-jclark, 2016-09-22:
I've experienced the same issue, and I've not found a way around it yet. Anyone track down the change that causes this failure?

j-rivero, 2016-09-23:
Umm interesting. Have you tried to do the same with any other node different that Gazebo? Is that working?

jaredmoore, 2016-09-23:
Yes, strangely enough if I spool up many different nodes, it is only the Gazebo node that fails to remap.  I can provide more code or a different example if needed.

jonbinney, 2016-11-30:
This commit in the kinetic-devel branch looks suspicious: https://github.com/ros-simulation/gazebo_ros_pkgs/commit/a90e609a81702b13bee235b079081edf68ff6971

jonbinney, 2016-11-30:
Here's the corresponding PR: https://github.com/ros-simulation/gazebo_ros_pkgs/issues/387

Sounds like remappings were intentionally removed; @scpeters @peci1 

Seems to me that when run through roslaunch/rosrun, remappings should be used.

j-rivero, 2016-12-01:
Thanks John for the commenting here. @jaredmoore the fix is now if the `kinetic-devel` branch, could you please check if that fixes your problem?

jaredmoore, 2016-12-01:
Sure thing, it might be a week or two as I'm in the middle of finals, but I will try and set some time aside.  Thanks for the digging on this!

jaredmoore, 2016-12-02:
I tested this on my installation after removing the apt-get acquired ros-gazebo-pkgs and building from the source on the devel branch.  The fix appears to work correctly and I am now able to remap the Gazebo topic.  It will be a few weeks before I can test it on my larger instance, but for now I am confident that this addresses the issue.  Thank you all for your help!

closed 2016-12-02 by jaredmoore