ros-simulation/gazebo_ros_pkgs #488: Crashes on zero value collision geometries

pabloinigoblasco, 2016-09-09:
I have written this issue on ros.answers.

I get one weird error trying to load some of the samples/models provided in gazebo_plugins on Ubuntu 16.04 with ROS kinetic installed via official repositories.

Particularly, for the multi_robot_scenario fails. On the other hand the tricycle demo or the pendulum demos work properly.

This is the error I get trying to load the multi_robot_scenario demo.

Error [Param.cc:451] Unable to set value [-nan -nan -nan] for key[size]
gzserver: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreNode.cpp:630: virtual void Ogre::Node::setScale(const Ogre::Vector3&): Assertion `!inScale.isNaN() && "Invalid vector supplied as parameter"' failed.
[r1/urdf_spawner-4] process has finished cleanly
log file: /home/geus/.ros/log/37a5e414-7527-11e6-8e8a-d8cb8abfd844/r1-urdf_spawner-4*.log
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
Aborted (core dumped)
[gazebo-2] process has died [pid 19670, exit code 134, cmd /media/geus/Storage/simulation_gazebo/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -e ode worlds/empty.world __name:=gazebo __log:=/home/geus/.ros/log/37a5e414-7527-11e6-8e8a-d8cb8abfd844/gazebo-2.log].
log file: /home/geus/.ros/log/37a5e414-7527-11e6-8e8a-d8cb8abfd844/gazebo-2*.log
In order to reproduce this issue you have to install the package ros-kinetic-gazebo-plugins and execute the launch file:

roslaunch $(roslaunch $(rospack find gazebo_plugins)/test/multi_robot_scenario/launch/multi_robot_scenario.launch

pabloinigoblasco, 2016-09-09:
It looks like there is a bug in gazebo 7.0.0 contained in the kinetic debian package repository for ubuntu 16.04. The bug appears when the urdf descriptions contains collision geometry objects with zero value parameters.

For instance in "gazebo_plugins/test/multi_robot_scenario/xacro/p3dx/pioneer3dx_wheel.xacro": we find this chunk of urdf:

<collision>
  <origin xyz="0 0 0" rpy="0 0 0"/>
  <geometry>
    <box size="0 0 0"/>
  </geometry>
</collision>
The box size must not be zero. The error is solved if the above code chunk is replaced by the code chunk below:

<collision>
  <origin xyz="0 0 0" rpy="0 0 0"/>
  <geometry>
    <box size="0.1 0.1 0.1"/>
  </geometry>
</collision>
It is needed to apply this correction in several place in order to make work this demo:

roslaunch $(rospack find gazebo_plugins)/test/multi_robot_scenario/launch/multi_robot_scenario.launch

andre-nguyen, 2016-12-19:
Can confirm, wasted a good 5 hours on this :cry:

j-rivero, 2016-12-19:
Thanks Pablo, Andre for reporting.

Is this error happening in newer versions of gazebo, i.e: current Gazebo-7.4.0? @pabloinigoblasco would you mind to send the patches you made to get the thing fixed?

andre-nguyen, 2016-12-19:
I'm running gazebo with whatever version was installed with ros kinetic

```
andre@desktop:~$ gazebo --version
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

```

jonbinney, 2016-12-19:
This isn't a bug in gazebo so much as a bug in `pioneer3dx_wheel.xacro`, right? I wouldn't expect zero size boxes to work, although there could probably be a better error message along the lines of "Collision objects cannot have zero size" .

andre-nguyen, 2016-12-19:
I can't comment on whether it's right or not but it looks like it's something that used to work so people did wrote it in their models.

In my case I hit the bug trying to load a mico arm model from https://github.com/Kinovarobotics/kinova-ros

jonbinney, 2016-12-19:
Based on the usage in the kinova urdf, what do you think the behavior should be? Is a zero-size box a point for the purposes of collision? Or is it truly empty collision geometry which never collides with anything?

andre-nguyen, 2016-12-19:
It's used in a virtual link https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_description/urdf/kinova_common.xacro#L52

```xml
    <xacro:macro name="kinova_virtual_link" params="link_name">
        <link name="${link_name}">
            <visual>
                <geometry>
                    <box size = "0 0 0"/>
                </geometry>
            </visual>
            <collision>
                <geometry>
                    <box size = "0 0 0"/>
                </geometry>
            </collision>
        </link>
    </xacro:macro>
```

So I suppose it's the latter and truly an empty collision geometry.

jonbinney, 2016-12-19:
Since it's used that way in existing urdfs, it sounds like it would make sense to add a special cases that notices zero-size collision objects, treats them as no collision geometry, and outputs a warning such as "Warning: ignoring zero-size collision geometry for link <link>". I'm all for not breaking existing urdfs. Not sure where exactly this code needs to go though.

jonbinney, 2016-12-19:
I suspect that the zero size collision boxes were added as a workaround to an old bug. There was a time a couple years ago when links without collision geometry would crash gazebo/gazebo_ros, and I remember adding zero size collision geometry to my own robots at the time. Those workarounds should (I think) be removed now, but giving a warning and having things work will give people time to update their URDFs.

j-rivero, 2017-01-04:
The problem is not strictly related to ROS but to gazebo itself. I was able to reproduce it using the `model.sdf`[1] file at the end of this comment.

The gdb backtrace points to Visual.cc file. During my debugging I discovered that the current gazebo7 branch does not suffer from the Core dump so according with this, there should be something in the [list of changes from gazebo7 branch to gazebo-7.4.0](https://bitbucket.org/osrf/gazebo/compare/gazebo7%0Dgazebo7_7.4.0) that has modified the problem, I was not able to discover exactly which commit.

Given the number of modifications pending it is my intention to release a new 7.5.0. If anyone want to continue debugging previous versions or improving error messages, I was working with the following patch (although not sure if it fixes the problem or it is fully correct):

```patch
diff -r ced7f59587e3 gazebo/rendering/Visual.cc
--- a/gazebo/rendering/Visual.cc	Thu Dec 22 19:43:25 2016 +0100
+++ b/gazebo/rendering/Visual.cc	Sat Dec 24 00:56:09 2016 +0100
@@ -2583,7 +2583,15 @@
     else
       gzerr << "Unknown geometry type[" << _msg->geometry().type() << "]\n";
 
-    this->SetScale(geomScale * this->dataPtr->scale / this->DerivedScale());
+    // Set scale if the geometry is not zero
+    if (geomScale == ignition::math::Vector3d::Zero)
+    {
+      gzerr << "Geometry is 0,0,0. Skipping scaling"  << std::endl;
+    }
+    else
+    {
+      this->SetScale(geomScale * this->dataPtr->scale / this->DerivedScale());
+    }
   }
 
   if (_msg->has_material())
```

[1] `model.sdf` file (to use it, create a new model in `~/.gazebo/models/` and write a minimal .config file. Insert it from the gazebo GUI).

```xml
<sdf version='1.6'>
  <model name='r1'>
    <link name='base_link'>
      <pose frame=''>0 0 0 0 -0 0</pose>
      <inertial>
        <pose frame=''>-0.0045 0 0.0234 0 -0 0</pose>
        <mass>10</mass>
        <inertia>
          <ixx>0.136322</ixx>
          <ixy>0</ixy>
          <ixz>0.009477</ixz>
          <iyy>0.189113</iyy>
          <iyz>0</iyz>
          <izz>0.178258</izz>
        </inertia>
      </inertial>
      <collision name='base_link_fixed_joint_lump__chassis_collision'>
        <pose frame=''>-0.045 0 0.148 0 -0 0</pose>
        <geometry>
          <mesh>
            <scale>1 1 1</scale>
            <uri>/home/jrivero/code/ros/ws/gazebo_ros_pkgs/src/gazebo_ros_pkgs/gazebo_plugins/test/multi_robot_scenario/meshes/p3dx/chassis.stl</uri>
          </mesh>
        </geometry>
        <surface>
          <contact>
            <ode/>
          </contact>
          <friction>
            <ode/>
          </friction>
        </surface>
      </collision>
      <collision name='base_link_fixed_joint_lump__top_collision_1'>
        <pose frame=''>-0.045 0 0.234 0 -0 0</pose>
        <geometry>
          <mesh>
            <scale>1 1 1</scale>
[... 110 lines cut ...]
      </visual>
      <velocity_decay/>
      <gravity>1</gravity>
      <velocity_decay/>
    </link>
    <joint name='right_hub_joint' type='revolute'>
      <child>right_hub</child>
      <parent>base_link</parent>
      <axis>
        <xyz>0 1 0</xyz>
        <limit>
          <lower>-1e+16</lower>
          <upper>1e+16</upper>
        </limit>
        <dynamics>
          <spring_reference>0</spring_reference>
          <spring_stiffness>0</spring_stiffness>
        </dynamics>
        <use_parent_model_frame>1</use_parent_model_frame>
      </axis>
    </joint>
    <pose frame=''>0 -0.5 0.01 0 -0 1.57</pose>
  </model>
</sdf>
```

j-rivero, 2017-01-12:
The new gazebo7 version 7.5.0 does not crash the simulator in my system anymore.

MahdiehNejati, 2017-06-19:
I'm having this problem in gazebo8 with ROS kinetic. Is there a way to fix this without reverting to an older version of gazebo? Thanks!

andre-nguyen, 2017-06-19:
I ended up adding a very small non-zero collision size to my model to fix the problem:

https://github.com/MRASL/kinova-ros/commit/2400703eadddd1f5ce2b24ae9fa600130f9d4bf5

ironmig, 2018-06-15:
I have confirmed on fresh installs of kinetic with gazeo7 and gazebo8 that this is not an issue. Please update to newer versions of gazebo is this issue remains, as it was patched

closed 2018-06-15 by ironmig

mhwasil, 2018-08-18:
Changing size values to non zero ones works for me