ros-simulation/gazebo_ros_pkgs #489: No ability to "step" the simulator

el3ment, 2016-09-09:
world_control messages are not exposed via the ros API

jaredmoore, 2016-09-12:
It is possible to step the simulator explicitly, but you need to write a gazebo plugin and ROS service.  Connect the two that way and make your service offer a step() method.  Make sure to confirm the step has taken place on the ROS side before moving on or your simulation and ROS code will diverge significantly.  I have a blog post available to explain this but I'm not sure what the linking policy here is.

j-rivero, 2016-09-12:
> I have a blog post available to explain this but I'm not sure what the linking policy here is.

There is no formal linking policy but please go ahead with the link if it provides relevant information about the issue topic.

jaredmoore, 2016-09-12:
In that case, it's actually three separate short posts.  I'd recommend starting with the first one, although you can jump around if needed.  

[First Post Here](http://jaredmmoore.com/ros-gazebo-for-evolutionary-robotics-stepping-the-world-pt-1/)

I do not have the confirmation message or service declared here.  In this case, it is strictly an example of how to get to the world_control topic with a ROS node.  I am planning to do a full write up of my solution in the future, although it is down the task list.