ros-simulation/gazebo_ros_pkgs #490: Gazebo is setting "sensor's last update time" to 0

StephanOpfer, 2016-09-14:
Hi all,

I am not sure, whether this issue is a problem of gazebo or the plugin, but please find here my repost of an [issue](https://bitbucket.org/osrf/gazebo/issues/2048/gazebo-is-setting-sensors-last-update-time), I already posted for the gazebo bitbucket tracker thing.

--

I found this issue, using ROS, AMCL, TurtleBot Simulation etc. I traced it back to line 191:

    this->depth_sensor_update_time_ = this->parentSensor->LastUpdateTime();

in [gazebo_ros_openni_kinect.cpp](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_openni_kinect.cpp). The parentSensor->LastUpdateTime() is 0 at 50% of the time.

I am using Ubuntu 16.04 LTS and ROS Kinetic with corresponding Gazebo Version 7.0.0. Here is how I reproduced the bug:

    roslaunch turtlebot_gazebo turtlebot_world.launch

This starts gazebo with one turtlebot. For the kinect plugin to send data, you need to subscribe to its ROS topic, because it uses lazy publishing:

    rostopic echo /camera/depth/image_raw/header

This subscribes only to the header of the image message. There you should see, if you can reproduce my issue, that the time stamp is 0 in ~50% of the time. I would expect it to be always the simulation time.

I hope someone could help here, because I didn't want to setup the whole gazebo project from source, in order to further debug it.

Greetings,
  Stephan

guihomework, 2016-09-15:
Hi, 
@StephanOpfer this is coincidence. I happen to notice two days ago and wanted to report today after more tests that quite often no time (0sec 0nsec) or maybe ~50% of the time too did not test so often, appears in the header of the Bumpers (ContactState) message. I had one gazebo launch when time was perfectly set ! So this is very annoying when it is not 100% reproducible.

I looked deep in gazebo code, version 2.2.3 (still running indigo here), and it looks like everything is okay in the Sensor LastMeasuredTime to be stored in the protobuf message (https://bitbucket.org/osrf/gazebo/src/177d54e5025e8ebd0af19ccbfb602ceeb0d8fc54/gazebo/sensors/ContactSensor.cc?at=gazebo-current_2.2.3&fileviewer=file-view-default#ContactSensor.cc-195) . Even the topic visualization in gazebo directly shows correct time. 
but on the plugin side, the time is empty for contacts, is filled for contact (single contact inside contacts) and is filled in lastmeasuredtime. I don't understand why.

I added some debug in the bumper
```
  common::Time gazebotime = this->world_->GetSimTime();
  ROS_DEBUG_STREAM("sim time " << gazebotime.sec << "," <<gazebotime.nsec);
  common::Time meastime = this->parentSensor->GetLastMeasurementTime();
  ROS_DEBUG_STREAM("measurement time " << meastime.sec << "," <<meastime.nsec);
  msgs::Time contacts_time = contacts.time();
  ROS_DEBUG_STREAM("contacts time via msgs " << contacts_time.sec() << "," <<contacts_time.nsec());
  ROS_DEBUG_STREAM("global contact time " << contacts.time().sec() << "," <<contacts.time().nsec());
  contact_time = ros::Time(contacts.time().sec(), contacts.time().nsec());

```
+ a DEBUG_STREAM inside the loop for each local contact
which outputs
```
[DEBUG] [1473794731.561772602, 28.015000000]: sim time 28,15000000
[DEBUG] [1473794731.561801712, 28.015000000]: measurement time 28,15000000
[DEBUG] [1473794731.561819986, 28.015000000]: contacts time via msgs 0,0
[DEBUG] [1473794731.561835947, 28.015000000]: global contact time 0,0
[DEBUG] [1473794731.561855423, 28.015000000]: local contact time 28,4000000
[DEBUG] [1473794731.561879138, 28.015000000]: local contact time 28,5000000
[DEBUG] [1473794731.561898337, 28.015000000]: local contact time 28,6000000
[DEBUG] [1473794731.561918832, 28.015000000]: local contact time 28,7000000
[DEBUG] [1473794731.561939278, 28.015000000]: local contact time 28,8000000
[DEBUG] [1473794731.561961089, 28.015000000]: local contact time 28,9000000
[DEBUG] [1473794731.561989262, 28.015000000]: local contact time 28,10000000
[DEBUG] [1473794731.562011669, 28.015000000]: local contact time 28,11000000
```

Tested yesterday on gazebo distributed with kinetic and same issue with the bumper.

ironmig, 2018-06-18:
I cannot replicate and this has likely been fixed by the PR's addressing #408. Comment to re open this issue if it still occurs on supported ROS distros.

closed 2018-06-18 by ironmig