ros-simulation/gazebo_ros_pkgs #491: [gazebo_ros] "headless" arg in launch files actually records

mrjogo, 2016-09-14:
Many of the gazebo_ros launch files expose an arg "headless" that is supposed to "Disable any function calls to simulator rendering (Ogre) components." (from http://gazebosim.org/tutorials?tut=ros_roslaunch), but actually sets the -r, or record, command line switch on gzserver. I expect that at some point that switch disabled rendering, but now there doesn't seem to be any such option (I'm using Gazebo 7.3.1):

```
$ gzserver --help
gzserver -- Run the Gazebo server.

`gzserver` [options] <world_file>

Gazebo server runs simulation and handles commandline options, starts a Master, runs World update and sensor generation loops.

Options:
  -v [ --version ]              Output version information.
  --verbose                     Increase the messages written to the terminal.
  -h [ --help ]                 Produce this help message.
  -u [ --pause ]                Start the server in a paused state.
  -e [ --physics ] arg          Specify a physics engine 
                                (ode|bullet|dart|simbody).
  -p [ --play ] arg             Play a log file.
  -r [ --record ]               Record state data.
  --record_encoding arg (=zlib) Compression encoding format for log data 
                                (zlib|bz2|txt).
  --record_path arg             Absolute path in which to store state data
  --seed arg                    Start with a given random number seed.
  --iters arg                   Number of iterations to simulate.
  --minimal_comms               Reduce the TCP/IP traffic output by gzserver
  -s [ --server-plugin ] arg    Load a plugin.
  -o [ --profile ] arg          Physics preset profile name from the options in
                                the world file.
```

jliviero, 2016-10-24:
+1, I have noticed the same (was about to post this same issue!).  I just ran a quick test to set `headless` and confirmed that it starts recording instead.  An example of the launch arg in this package [is here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_ros/launch/empty_world.launch#L26), and the conflicting Gazebo code (well, help text) [is here](https://bitbucket.org/osrf/gazebo/src/192a57657ec0d4873f695a3dec026064aec0240e/gazebo/gazebo_main.cc?at=default&fileviewer=file-view-default#gazebo_main.cc-54).  It looks like this change has been backported all the way to [Gazebo 2.0](https://bitbucket.org/osrf/gazebo/src/436b33b931e58bdcbbe2f6290df464117d529771/gazebo/gazebo_main.cc?at=gazebo_2.0.0&fileviewer=file-view-default#gazebo_main.cc-49).

j-rivero, 2016-10-31:
I don't think that we currently have any option to disable rendering in upstream gazebo so makes totally sense to me to remove this option. Thanks for the issue.

j-rivero, 2016-11-01:
Related pull request to fix documentation: https://bitbucket.org/osrf/gazebo_tutorials/pull-requests/379/rename-headless-parameter-to-recording/diff

ironmig, 2018-05-30:
Looks like this was patched, can close.

closed 2018-06-15 by ironmig