ros-simulation/gazebo_ros_pkgs #508: Namespace change does not work in gazebo

talkingjeff, 2016-11-12:
When I provided a namespace when launch gzserver, it will always say 'waiting for /gazebo/set_physics_properties'. It should have been waiting for /my_namespace/set_physics_properties instead.

The following could solve the issue:
In method GazeboRosApiPlugin::physicsReconfigureThread, do these changes:
-  physics_reconfigure_set_client_ = nh_->serviceClient<gazebo_msgs::SetPhysicsProperties>("/gazebo/set_physics_properties");
-  physics_reconfigure_get_client_ = nh_->serviceClient<gazebo_msgs::GetPhysicsProperties>("/gazebo/get_physics_properties");
+  physics_reconfigure_set_client_ = nh_->serviceClient<gazebo_msgs::SetPhysicsProperties>("set_physics_properties");
+  physics_reconfigure_get_client_ = nh_->serviceClient<gazebo_msgs::GetPhysicsProperties>("get_physics_properties");

j-rivero, 2016-12-07:
Thanks for the issue, some questions:

* Which version of ROS/Gazebo are you using?
* Can you provide a reproducible use case?
* Could you open a PR with the changes or attached them using the `git diff` or any other diff tool.

Thanks.

talkingjeff, 2016-12-08:
Hello,

I encountered this issue for both Indigo/Gazebo 7.4.0 and Kinetic/Gazebo
7.0.0.

I attached a launch file that would try to launch gzserver with namespace
'client_000' and node name 'gzserver'.
It could not because gazebo_ros_api_plugin would wait for
service [/gazebo/set_physics_properties] while only service
[/client_000/gzserver/set_physics_properties] is provided.

Make the changes as in the attached git diff log would solve the issue.

Thanks,
Jun

-- 
Jun Dong
Department of Computer Science
Rensselaer Polytechnic Institute
talkingjeff@gmail.com

diff --git a/gazebo_ros/src/gazebo_ros_api_plugin.cpp b/gazebo_ros/src/gazebo_ros_api_plugin.cpp
index 4310a4a..f337620 100644
--- a/gazebo_ros/src/gazebo_ros_api_plugin.cpp
+++ b/gazebo_ros/src/gazebo_ros_api_plugin.cpp
@@ -1965,8 +1965,8 @@ void GazeboRosApiPlugin::physicsReconfigureCallback(gazebo_ros::PhysicsConfig &c
 
 void GazeboRosApiPlugin::physicsReconfigureThread()
 {
-  physics_reconfigure_set_client_ = nh_->serviceClient<gazebo_msgs::SetPhysicsProperties>("/gazebo/set_physics_properties");
-  physics_reconfigure_get_client_ = nh_->serviceClient<gazebo_msgs::GetPhysicsProperties>("/gazebo/get_physics_properties");
+  physics_reconfigure_set_client_ = nh_->serviceClient<gazebo_msgs::SetPhysicsProperties>("set_physics_properties");
+  physics_reconfigure_get_client_ = nh_->serviceClient<gazebo_msgs::GetPhysicsProperties>("get_physics_properties");

...

ironmig, 2018-05-29:
Duplicate of #507.