ros-simulation/gazebo_ros_pkgs #512: Port gazebo_ros_pkgs to ROS 2.0

j-rivero, 2016-11-29:
Interest into porting `gazebo_ros_pkgs` to ROS 2.0 is [starting to appear within the ROS Community](http://discourse.ros.org/t/questions-related-to-gazebo-turtlebot-supported-in-ros-2/830), this post tries to server a tracker for suggestion, status and discussions on how to do it.

**Update 29 Nov 2016**
The ROS team currently warns about implementations at this point since the API is still unstable an future changes could require a big amount of work to update the code. Even with API instability a possible suggestion to start the works could be some time [after the Beta1 (release planned in the next weeks)](https://github.com/ros2/ros2/wiki/Roadmap), late January if the feedback regarding to the Beta1 is good at that moment.

jpgr87, 2017-08-17:
ROS 2 1.0 is currently slated to be released sometime before the beginning of next year - is gazebo working towards support in that timeframe?  There's been some interest in ROS 2 support for gazebo early next year, but I don't see it anywhere on gazebo's roadmap.

CC @bponsler

j-rivero, 2017-08-17:
> ROS 2 1.0 is currently slated to be released sometime before the beginning of next year - is gazebo working towards support in that timeframe? There's been some interest in ROS 2 support for gazebo early next year, but I don't see it anywhere on gazebo's roadmap.

As far as I know, there is no current plan at this moment to port `gazebo_ros_pkgs` to ROS 2 by the OSRF although some people, as you mention, has been doing some code to support it (the friends of Erle might have something working cc @vmayoral) . I did not receive any notification during these last months but many people tend far from the public scope.

vmayoral, 2017-08-18:
Hello @j-rivero! We're indeed interested in this feature but haven't put time into it just yet. For now, all the tests we've done have made use of the bridges.

j-rivero, 2018-06-02:
Update: the branch `ros2` (based on kinetic-devel) will be used to import the changes related to ROS2 port. The first pull request is #735.

ironmig, 2018-07-11:
I've been doing some work on this and have some ideas for the design I'd like to put to the community for discussion:

1. Port gazebo_msgs to ROS2 (see #735)
   * Add headers to 
1. Create a base class (GazeboRosNode?), which inherits from rclcpp::Node but also adds itself to a global executor (callback threads) to provide parity to the ROS1 implementation where plugins don't worry about spinning.
   * GazeboRosNode can optionally be given an SDF element for construction (like for Model/World plugins), which will configure [the Node](http://docs.ros2.org/bouncy/api/rclcpp/classrclcpp_1_1_node.html#ab28b0b3c08b1a3597b1e0442822f46f2) based on the sdf. This will address the confusion over namespaces and lack of remapping functionality in the current repo
   ```
    <!-- Optional configurations for a plugin's Node -->
    <ros>
      <!-- Name of node, only use if plugin is one node -->
      <node_name></node_name>
      <!-- Namespace of the node -->
      <namespace></namespace>
      <!-- Command line arguments sent to Node's contructor for remappings -->
      <arguments>
         <argument>my_topic:=new_topic</argument>
         <argument>__name:=super_cool_node</argument>
      </arguments>
      <!-- Initial ROS params set for node -->
      <parameters>
         <parameter name="max_velocity">55</parameter>
         <parameter name="publish_odom">True</parameter>
      </parameters>
    </ros>
   ```
1. Port gazebo_ros_api plugin (the one that does ModelStates, spawn model, etc), and initializes ROS
1. Port gazebo_ros_paths_plugin (which adds ROS packages to the Gazebo paths for models/textures/etc)
1. Port the scripts/launch files in gazebo_ros used to open gazebo with the above ROS plugins
1. Port the other most used plugins (Camera, IMU, laser, DepthCamera)
   * Note the Camera plugins will be blocked until image_transport is ported to ROS2
1. As plugins are ported over, follow [ROS2 style guidelines](https://github.com/ros2/ros2/wiki/Developer-Guide#c-1) including doxygen style commenting
1. Lowest priority is porting gazebo_ros_control, as ros_control is not present in ROS2

ironmig, 2018-07-13:
Implementations for (1) and (2), besides for the SDF configuration are in #769

chapulina, 2018-07-19:
The design and progress will be tracked on [this wiki page](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-integration).

mkhansen-intel, 2018-10-09:
Is there an update on this, it looks like ros2_control is done, (https://github.com/ros-controls/ros2_control/issues/1) so gazebo_ros_control should be unblocked, right?

chapulina, 2018-10-09:
> Is there an update on this, it looks like ros2_control is done, ([ros-controls/ros2_control#1](https://github.com/ros-controls/ros2_control/issues/1)) so gazebo_ros_control should be unblocked, right?

There's currently no plan to port `gazebo_ros_control`. A pre-requisite would be to get `ros_control` into [ros2.repos](https://github.com/ros2/ros2/blob/master/ros2.repos).

bmagyar, 2018-11-12:
I currently don't have time to maintain `ros2_control`. I'd be happy to welcome a maintainer onboard for moving the current proof of concept forward which should then enable proper gazebo integration.

chapulina, 2018-12-08:
Added a wiki page with a few guidelines for peeps interested in helping migrating plugins to ROS 2: https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-migration-contribution-guide

DhruvKoolRajamani, 2019-03-12:
Redirected here from the GSoC osrf Wiki. Which plugins are still waiting to get ported?

chapulina, 2019-03-12:
@DhruvKoolRajamani , I just updated this issue above, as well as #779. Thanks for the interest!

mkhansen-intel, 2019-03-18:
Is there a status update on gazebo_ros_depth_camera? It doesn't look from this issue like it has been ported yet. If it has, great I'd like to use it.

chapulina, 2019-03-18:
> Is there a status update on gazebo_ros_depth_camera?

No work has started on it yet, AFAIK.

mkhansen-intel, 2019-03-19:
> No work has started on it yet, AFAIK.

Any ETA on when?

chapulina, 2019-03-19:
> Any ETA on when?

There's no one assigned to work on this right now. We may get a GSoC intern during the summer, but I could also review a PR if someone else works on it before that.

shiveshkhaitan, 2019-03-20:
I would like to work on this. Will try to get it done as soon as possible.

bmagyar, 2019-03-20:
I can't really dedicate any time to porting the `ros_control` plugin but would be happy to review one!

Yeshasvitvs, 2019-03-30:
@chapulina which is the current stable branch for `gazebo_ros_pkgs` ? 

I tried the branches `crystal` and `ros2` as dodumented [here](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-integration#clipboard-overview) against [ros2 master](https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos) but the `gazebo_ros_pkgs` could not be build.

chapulina, 2019-03-31:
Crystal:

* Crystal debs **or** [ros2.repos from the crystal branch](https://raw.githubusercontent.com/ros2/ros2/crystal/ros2.repos) + the correct branches from `image_common` and `vision_opencv`
* `gazebo_ros_pkgs` `crystal` branch

Dashing:

* `ros2.repos` master branch + [these extra repos](https://bitbucket.org/snippets/chapulina/geRKyA/ros2repos-supplement-gazebo_ros_pkgs)  (which include `gazebo_ros_pkgs`'s `ros2` branch)

Yeshasvitvs, 2019-03-31:
@chapulina Thanks for the update. 
It was a bit unclear with the current documentation available. I created [installation notes](https://gist.github.com/Yeshasvitvs/5f7cf2cac48fbe70dac399f68791e6fa) that may help developers of  GsoC2019 to get started with setting up the code base. Please review it and let me know if you notice some mistakes.

chapulina, 2019-04-01:
@Yeshasvitvs , that all looks right for me, thanks for taking the time to write it. You're welcome to add the detailed instructions to the [installation tutorial](http://gazebosim.org/tutorials?tut=ros2_installing&cat=connect_ros).

shivaang12, 2020-02-19:
@chapulina I read some where that `gazebo/get_world_properties` service will be ported in `gazebo_ros_properties`. But haven't seen this service in `eloquent` branch. Can you shed some light on it's update? Also, curious to know why `gazebo_ros_properties` is absent from `ros2` branch?

Thanks!

chapulina, 2020-02-19:
> I read some where that gazebo/get_world_properties service will be ported in gazebo_ros_properties

That seems to be missing on the [migration guide](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-gazebo_ros_api_plugin). The new service is now `get_model_list`, provided by the [gazebo_ros_factory](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/dashing/gazebo_ros/src/gazebo_ros_factory.cpp#L142) plugin.

> curious to know why gazebo_ros_properties is absent from ros2 branch?

I just made a pull request today forward-porting `gazebo_ros_properties` from `dashing` to `eloquent`, see #1049. Now someone needs to make a PR from `eloquent` to `ros2` (i.e. `foxy`).

guru-florida, 2020-09-27:
What is the latest plans for porting gazebo_ros_control? I see in the Roadmap repo that it is showing "Blocked by ros-controls/ros2_control#1" which I believe is ready, yes? Has any work started on it that I can jump into?

If ros2_control is ready, briefly, what work would remain to complete the port? 

- Derive from ros_gazebo/Node
- move existing code from ROS1 gazebo_ros_control over and convert the rclcpp pub/sub setup/callbacks
- Are there any special simulated parts of ROS1 gazebo_ros_control that depend on parts of ros_control that have been overhauled? Such as trajectory computation, trapezoidal stuff, effort, position, velocity controllers, etc?

chapulina, 2020-09-28:
@guru-florida , I updated the table with a link to the repository where that work is happening: https://github.com/ros-simulation/gazebo_ros2_control

guru-florida, 2020-09-28:
Thanks @chapulina! I can't believe I didnt find that repo, it was right beside the gazebo_ros_pkgs..how!?!?! I'm blind. lol. I will check it out. :+1:

chapulina, 2021-02-19:
With the release of `gazebo_ros2_control` into Foxy, I believe this issue can be closed :tada:

closed 2021-02-19 by chapulina